sensorium/Mozzi #166: EXTERNAL_AUDIO_OUTPUT not working on RP2040?

tomcombriat, 2023-01-31:
Apparently, according to: https://groups.google.com/g/mozzi-users/c/csMUSbZjnf8

Will try to reproduce.

tomcombriat, 2023-01-31:
I confirm this bug, at least when trying to use I2S via an external audio output function.
Weird things:
 - an analogWrite in `audioOutput` seems to work (PWM frequency is to low thought), does not hang
 - putting the I2S output function in `updateAudio` works (but not clean as this function can be called in burst)
 - putting the I2S output function in a user defined `audioOutput` works for approx 0.5s, then makes weird signal (random) and then hang the whole chip. The weird signal actually looks like what we got when the conflict on IRQ with the I2S was not solved yet.
I have to say that I really do not have any idea of why this is happening… Especially as the I2S (did not try SPI) works in `updateAudio` I kind of rule out an I2S problem, why putting it in `audioOutput` changes something?

Any idea?

tfry-git, 2023-01-31:
So what you tested is essentially that you took the code for EXTERNAL_DAC_VIA_I2S but placed it in a custom audioOutput(), instead? And in that case is does not work? Weird.

One idea: Could it be that the speed of the I2S does not quite match the rate at which audioOutput() gets called? Perhaps the buffer is getting overcrowed, eventually resulting in a hang? If that is the issue, checking ```i2s.availableForWrite()``` should help.

tomcombriat, 2023-01-31:
> So what you tested is essentially that you took the code for EXTERNAL_DAC_VIA_I2S but placed it in a custom audioOutput(), instead? And in that case is does not work? Weird.

Yup.

I do not really care about getting it to work with I2S (as this is already implemented in "hardware") but there was a report of a similar problem on the forum with I2C, which triggered me to test the `EXTERNAL_AUDIO_OUTPUT` on RP2040 (did you try?).
 That being said, I2C is usually too slow to sustain an `AUDIO_RATE`, so it might be the same problem. Will try one of these days with SPI emulating I2S (without buffer) to see if this error remains.

tomcombriat, 2023-02-01:
Okay, using `i2s.availableForWrite()` as a crude test seems to work (with a few glitches because of the two buffers probably not running exactly at the same time, that is not an optimal situation).

So closing this for now.

closed 2023-02-01 by tomcombriat

tfry-git, 2023-02-05:
Ok, actually, I think setting BYPASS_MOZZI_OUTPUT_BUFFER is the "clean" way to go in this situation, then. That makes sense, too, as the whole trouble is that there already is another buffer at work.

tomcombriat, 2023-02-05:
Yeah, but I think there is no need to have this work cleanly with I2S as it is ported already. This was just a mockup test to see if the `EXTERNAL_AUDIO` was working (and I had this DAC wired on my breadboard ;) ).

tfry-git, 2023-02-05:
Sure, I was just saying this for completeness (and because I had sort of forgotten that we already have a ready solution for dealing with the case of an independent output buffer somewhere).