sensorium/Mozzi #168: Hardware support for the PicoADK RP2040 DSP Board

DatanoiseTV, 2023-02-07:
The PicoADK is a board which is focused on building standalone synthesizers, oscillators etc.
By default, it uses the PicoADK-Firmware-Template, which is based on the Pico SDK.

It has a 32-bit Audio Codec and a 8 channel 12-bit 3.3V/5V ADC with up to 1 Megasample.

Would be cool to see Mozzi support for it to make it also useful for Arduino IDE users.
https://github.com/DatanoiseTV/PicoADK-Hardware
https://github.com/DatanoiseTV/PicoADK-Firmware-Template

tomcombriat, 2023-02-08:
Hi,

I actually wonder if this should not be working already, probably with minor configuration:
 - the DAC is I2S, which has been ported #163 
 - the ADC is SPI which is compatible with Mozzi (probably with a restriction that Audio input should be on one of the bare chip ADC pins and not the external ADC).

Have you tried :) ?

DatanoiseTV, 2023-02-08:
I didn't figure out yet where to start. Is Mozzi able to handle 16-bit/32-bit I2S output?

tomcombriat, 2023-02-08:
Yup. First step I think would be to adapt "AudioConfigRP2040.h" to the hardware:
 - the [specific pins ](https://github.com/sensorium/Mozzi/blob/5925038c2ebe7f4cc0f8d56edebc8a05eae197ca/AudioConfigRP2040.h#L31-L33) used for the DAC: 
 - the [number of bits](https://github.com/sensorium/Mozzi/blob/5925038c2ebe7f4cc0f8d56edebc8a05eae197ca/AudioConfigRP2040.h#L36) used.
 - this DAC seems to accept both left justified and standard I2S, so as long as you set its config pin correctly you can choose any mode.

Then I would just try to run one of the examples

DatanoiseTV, 2023-02-13:
@tomcombriat Would you be able to help out with it? The board has been merged in the "arduino-pico" package so that board which have I2S pins defined will be working out-of-the-box.
https://github.com/earlephilhower/arduino-pico/blob/master/libraries/I2S/src/I2S.cpp#L33

DatanoiseTV, 2023-02-13:
@tomcombriat https://github.com/sensorium/Mozzi/pull/169
I've also added random seed support using the ROSC of the RP2040. This is the way
recommended by Raspberry.

tomcombriat, 2023-02-13:
I made a few review on your PR.
For the DAC I do not have one to test this, but as I said in the review I do not think this level of details should go in Mozzi but probably more in your sketches (or any higher level api that goes with the board).
Let's see what the others say.

DatanoiseTV, 2023-02-13:
I will review it. Yes, I've tested using all the examples and it works.
The reason for adding the PicoADK specific mute pin control is that the user won't need to modify sketches in order to use a PicoADK.

pschatzmann, 2023-02-28:
In the config we can define an EXTERNAL_AUDIO_OUTPUT. I think it would be cool to have the same on the input side, so that we could add any custom audio input in the sketch itself. 

This would also help e.g. for the ESP32 where the getADCReading() is not implemented.