sensorium/Mozzi #170: rPi Pico RP2040 Documentation

PeteHaughie, 2023-02-16:
I realise that the RP2040 is a new addition to the supported boards but I'm finding it hard to get my head around how to go about writing sketches and circuits for the board.

Is the default the PWM pin the same? Is HIFI mode necessary and supported? Do we need an external DAC?

An example sketch and layout in an article explaining the differences would be greatly appreciated.

PeteHaughie, 2023-02-16:
I've gone back and searched through the docs and noticed that the audio output pin is set to default GP0 which is essentially A0. So for anyone needing this info:
Is the default the PWM pin the same? No
Is HIFI mode necessary and supported? No and no
Do we need an external DAC? No

tomcombriat, 2023-02-16:
Hi Pete,

The README file is up-to-date on the RP2040, with the default pin specified. The "Hardware specific notes" also tell the different modes available for this platform and how to configure. The website is always a bit behind as it is not regenerated every time something is added.
If you have some suggestions about how to enhance the documentation do not hesitate :)!

Will close this issue soon as the problem is solved.

pschatzmann, 2023-02-28:
I had a look at the Readme and was positively surprised that an external DAC is supported now as well.

closed 2023-02-28 by tomcombriat