sensorium/Mozzi #171: got error becuse of name file

ff07aa, 2023-02-20:
In file included from C:\Users\jibin\Documents\Arduino\helios4_6\helios4_6.ino:108:0:
C:\Users\jibin\Documents\Arduino\libraries\Mozzi-master/StateVariable.h:53:6: error: multiple definition of 'enum filter_types'
 enum filter_types { LOWPASS, BANDPASS, HIGHPASS, NOTCH };
      ^~~~~~~~~~~~
In file included from C:\Users\jibin\Documents\Arduino\libraries\Mozzi-master/LowPassFilter.h:16:0,
                 from C:\Users\jibin\Documents\Arduino\helios4_6\helios4_6.ino:100:
C:\Users\jibin\Documents\Arduino\libraries\Mozzi-master/ResonantFilter.h:71:6: note: previous definition here
 enum filter_types { LOWPASS, BANDPASS, HIGHPASS, NOTCH };
      ^~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

tomcombriat, 2023-02-20:
Ah, indeed, did not tested to put both the StateVariable and the ResonantFilter in the same sketch. That makes sense.
Will commit a fix soon (tonight hopefully).
In the meantime, just use only one of them ;)

tomcombriat, 2023-02-22:
Hei @ff07aa,
A fix has just been merged in the library, please report if the problem persists!

PR #172 (2023-02-22): Fix duplicated enum between StateVariable and ResonantFilter
This fixes #171.
@tfry-git and @sensorium if you comments on the naming please let me know.

Otherwise I'll merge it soon!

Cheers,
Tom
closed 2023-02-22 by tomcombriat