sensorium/Mozzi #174: multiline2.h non-functional?

jepyang, 2023-02-28:
pretty keen to use multiline2.h for a more complex envelope project, but having trouble getting any results. I tried porting basically the same code over to ADSR.h and things suddenly sprung to life. I'm not discounting whatsoever the possibility that I am just using multiline2 incorrectly.  but there's no example code, and there's seemingly no mention anywhere on the internet of anyone actually using multiline2 (or multiline, for that matter)—the only references I can find by searching are the documentation itself. So I wanted to go ahead and ask here purely as a sanity check before I get much further.

will post some example code that seems like it should work but doesn't really.

sensorium, 2023-03-01:
Hi jepyang,
multiline and multiline2 are leftover work in progress from when I was mostly writing Mozzi for my own use.  They shouldn't really be in the master branch, so apologies for my poor housekeeping.  Maybe we can get something working if it looks like it could be useful.

jepyang, 2023-03-01:
Multiline2 is definitely interesting, but don’t go to any trouble on my account. :) Just glad to know I’m not imagining things!

closed 2023-03-01 by jepyang