sensorium/Mozzi #175: Make new release

per1234, 2023-03-07:
Arduino would like to add this valuable library to the [Library Manager](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-installing-a-library).

I understand that [the current recommendation](https://github.com/sensorium/Mozzi#installation) is for the users to manually download and install the development version of the library. However, Library Manager would make it easier for users to install and update the library and for developers to manage it as a dependency.

Arduino Library Manager uses Git tags as the units of release and at the time of the last tag the library did not meet the [requirements for admission to Library Manager](https://github.com/arduino/library-registry/blob/main/FAQ.md#what-are-the-requirements-for-a-library-to-be-added-to-library-manager) due to the presence of a symlink so it is currently impossible to add the library to Library Manager. The symlink has been removed (https://github.com/sensorium/Mozzi/pull/156) since the time of the tag, so the development version of the library is fully compliant with the submission requirements.

The necessary actions:

1. Update the `version` field value in [the `library.properties` metadata file](https://github.com/sensorium/Mozzi/blob/master/library.properties).
1. Create a new [GitHub release](https://docs.github.com/en/github/administering-a-repository/releasing-projects-on-github/managing-releases-in-a-repository) or [tag](https://git-scm.com/docs/git-tag) in the repository.

tomcombriat, 2023-03-09:
Hi,
This can be done very soon, thanks for explaining the missing steps!

@sensorium I can try to take care of it if you want. Since a lot has happened since the previous release, I propose going to version number 1.1.0. We can also go to 1.0.4 if you prefer for continuity. 
Version name: *Aedes daliensis* ?

Best,

sensorium, 2023-03-09:
Hi there,

great Tom, if you and Thomas are ready to release a version.  1.1.0. is fine by me.

Before releasing though, I should do a couple of things... (give me a few days)
- remove some fluff like unfinished classes, etc.
- regenerate the documentation (unless you're set up to do it Tom?)

Do you think the html docs should still be included in the download?  They are the main part of it, but I don't know if many people use them locally...

Any special reason for choosing Aedes daliensis?  I'll go with it.

tomcombriat, 2023-03-09:
Hi again,

Sounds good! I do not have much experience with Doxygen so maybe I will let you do it. I personally like the fact that the doc is embedded which is very useful for fiddling without internet access. But that might be only me…
No reason for *daliensis*, I just liked the name, did not know if there was a logic in the Aedes genus before or not, that was just a proposition.

Best,

PS: I case you are up for testing, maybe #124 could make it in that release?

tfry-git, 2023-03-09:
>     * remove some fluff like unfinished classes, etc.

For some of the unfinished stuff (I'm thinking of MultiLine2, in particular), I think it may be worth keeping after all, but perhaps in a dedicated subdirectory "experimental"?

sensorium, 2023-03-15:
OK, now I have a branch for MultiLine2, till I get around to fixing it. 
 
I've deleted cogl_sqrti.h from the master, it didn't need to be part of Mozzi and would be very unlikely to be used by anyone.

I've regenerated the docs.  They can always do with more straightening out, but that's ongoing.. I'll copy them to the website.

We've got Tom's great Wavefolder and MetaOscil classes.

Anything else need attention before a release?  (I don't mind if we need to re-release now and again, it will always be imperfect).

I haven't got my full testing scripts and hardware organised (for a few years now), but there don't seem to be complaints at the moment about actual synthesis issues, so maybe the library is working in general...

tomcombriat, 2023-03-15:
Hi @sensorium ,
I think we try to be quite extensive with testing before putting new things here and a few fixs have been added recently, so I would say the library is in a good working condition!

Making releases from time to time might also help us to allow more experimental stuff on the github version and releasing them when fulling tested, which I guess would probably help the workflow of having new things in Mozzi.

sensorium, 2023-03-16:
Oh, and there's Tom's MultiResonantFilter too!  Docs are updated again after the recent tidy-ups.

sensorium, 2023-03-19:
Hi @tomcombriat, if you're ready then I think I've done the few things I meant to do before the release.  Is there anything obvious I've missed?

Oh, I can't see your MultiResonantFilter class, but the example is there..
I hope I haven't lost it amongst my git meanderings.  Any idea what's happened?
Also, I wasn't sure if you might want to put anything in extras/NEWS.txt about your recent work?

tomcombriat, 2023-03-19:
Hi @sensorium , I think it looks good! Hope that is okay that I have added a little bit of information on NEWS.txt but did not go into details about the new classes as examples are there for that.

---

> Oh, I can't see your MultiResonantFilter class,

`MultiResonantFilter` is inherits nearly completely from `ResonantFilter`, so I put them in the same file (there is no new code for the MultiResonant).

I think we are good to go! Are you taking care of actually pushing the release?

tomcombriat, 2023-03-20:
Also, maybe @tfry-git has some comments on the release.

tomcombriat, 2023-03-21:
> I believe you have been the primary contributor of new stuff, lately.

But there is a (little) bit of you in each of these PR ;). Also, the abstraction of the guts which helped a lot porting new platforms is not to be under estimated.

Alright, I will try to push the new release soon then, hopefully tomorrow!

sensorium, 2023-03-21:
I can do the release, Tom.  Sorry, I should have responded earlier.

sensorium, 2023-03-22:
OK, done. Fingers crossed.

per1234, 2023-03-22:
Thanks everyone!

I have added the library to Arduino Library Manager. It takes a little while for the index to be regenerated and for the updated index to propagate through the cache to all users, but it should be available for installation via Library Manager within a day's time.

closed 2023-03-22 by per1234