sensorium/Mozzi #178: Arduino giga port

tomcombriat, 2023-04-03:
Hi,
This is just to trigger some conversations. Got hand on a giga from Arduino and was thinking to try to port it soon. I know that @tfry-git is also probably working on it. So I was wondering what would be the best strategy? I think merging things from someone else on a PR is not very convenient. 

Maybe we should create a branch and then PR to this one, so that we do not have to care about breaking the lib while testing and everyone can easily PR (and get merged eventually) on this branch? Of course the branch would be destined in between deleted once things are in master.

What do you think?

tfry-git, 2023-04-03:
> Maybe we should create a branch and then PR to this one

Sounds like a good idea. I already started my port in a branch "devel/arduino-giga". Actually, I had intended that to be in my personal fork, but I messed up, and so that branch is here, already.

As for the state / coordination / first thoughts:

- Audio output on the internal DAC is working, already, although you will need the bleeding edge version of Arduino_AdvancedAnaloge, referenced in this issue report: https://github.com/arduino-libraries/Arduino_AdvancedAnalog/issues/35 .
- I tried adding a simple timer driven PWM output (the timer also being what is needed for EXTERNAL_AUDIO_OUTPUT), but getting a high-resolution PWM turned out to be not as easy as I had though. I have given up on that one for now. (However, my idea was that this mode would provide most flexibility on what pins to use, but also be most portable across MBed boards).
- A further mode that I had in mind was PDM-coded output through whatever hardware interface will allow this most easily (I believe UART looks easy).
- I have not really looked into analog input at all, yet.

I suppose PDM-coded via UART is the area that I would be most likely to experiment with over the next few days, personally.

tfry-git, 2023-04-03:
Oh, and from here on I will try to make sure to work in my own fork, first, and then PR!

tfry-git, 2023-04-03:
> Actually, making the developments here was what I have in mind as this repo is the middle point for contributors.

I see what you mean, now. Yes, let's make it so.