sensorium/Mozzi #187: getAudioInput not defined

Quinn1876, 2023-04-15:
Hello,

I am currently trying to run the `AudioInput` example for mozi, but it will not compile.

Here is the code for reference:

```.ino
/*  Test of audio input using Mozzi sonification library.

 An audio input using the range between 0 to 5V on analog pin A0
 is sampled and output on digital pin 9.

  Configuration: requires these lines in the Mozzi/mozzi_config.h file:
  #define USE_AUDIO_INPUT true
  #define AUDIO_INPUT_PIN 0

 Circuit:
 Audio cable centre wire on pin A0, outer shielding to Arduino Ground.
 Audio output on DAC/A14 on Teensy 3.0, 3.1, or digital pin 9 on a Uno or similar, or
 check the README or http://sensorium.github.io/Mozzi/

 		Mozzi help/discussion/announcements:
  https://groups.google.com/forum/#!forum/mozzi-users

  Tim Barrass 2013, CC by-nc-sa.
*/

#include <MozziGuts.h>

void setup(){
  startMozzi();
}

void updateControl(){
}

AudioOutput_t updateAudio(){
  int asig = getAudioInput(); // range 0-1023
  asig = asig - 512; // now range is -512 to 511
  // output range in STANDARD mode is -244 to 243,
  // we scale down only by one bit, so you might need
  // to adjust your signal to suit
  return MonoOutput::fromAlmostNBit(9, asig).clip();
}

void loop(){
  audioHook();
}

```

Compile Error: 
`error: 'getAudioInput' was not declared in this scope
   int asig = getAudioInput(); // range 0-1023`

I looked around, but I could not find documentation for the following line:
`  Configuration: requires these lines in the Mozzi/mozzi_config.h file: `

Which config file do I need to add these lines too? Do I need to make this file path in the sketch directory or is there a global config file that I was supposed to change. If so, where do I find it?

Any help in this matter would be appreciated.

tomcombriat, 2023-04-15:
Hi!

Indeed, Mozzi needs to be configured to use the `AudioInput` feature.
For that you need to edit the file `mozzi_config.h` :

`//#define USE_AUDIO_INPUT true` should be : `#define USE_AUDIO_INPUT true`

To find this file, open your sketchbook folder (open Arduino's preferences to find where it is if you do not know), then `libraries/Mozzi`.

tomcombriat, 2023-07-26:
Hi @Quinn1876!

 Did that help? Can this issue be closed?

closed 2023-07-26 by Quinn1876