sensorium/Mozzi #193: Stereo - Strange error compilation depending if a function was define before - Problem with preprocessor ?

BertheGrandPied, 2023-05-18:
Hello,

I got a really strange problem using the stereo mode on a pico using pt_8211 as output (I did not test on other platforms).

# The problem

My code is using a rotary encoder and mozzi and looks like that (minimal code example, I remove a max of things)  :

```c++
//********************************************************************
// 2023 Djevahirdjian Léo
//
// Bug Stereo - Minimal Code
//
//********************************************************************

//********************************************************************
//     DEFINITIONS - Interface
//********************************************************************
// Rotary encoder
# define ROT_PIN0 6
# define ROT_PIN1 7
#include <RotaryEncoder.h>
RotaryEncoder *interface_encodeurRotatif = nullptr ;

//  Definition of the function called when rottary encoder is used.
// THIS MAKE THE COMPILATION FAILS
void callback_updateEncodeurRotatif(){
  interface_encodeurRotatif->tick() ;
}

//********************************************************************
//     DEFINITIONS - Mozzi
//********************************************************************
#include <MozziGuts.h>
#include <AudioOutput.h>
// Template oscillateur
#include <Oscil.h>
// Chargement des tables pour oscillateurs
#include <tables/saw2048_int8.h>
#include <tables/sin2048_int8.h>

Oscil <SAW2048_NUM_CELLS, AUDIO_RATE> Oscillateur_Scie1(SAW2048_DATA) ;
Oscil <SAW2048_NUM_CELLS, AUDIO_RATE> Oscillateur_Scie2(SAW2048_DATA) ;

int CH1_frequence = 440 ;
int CH2_frequence = 220 ;

void updateControl() {
  Oscillateur_Scie1.setFreq(CH1_frequence) ;
  Oscillateur_Scie2.setFreq(CH2_frequence) ;
}

// Fonction de génération de la sortie
AudioOutput_t updateAudio() {
  return StereoOutput::fromNBit(9, Oscillateur_Scie1.next(), Oscillateur_Scie2.next()) ;
}

//********************************************************************
//     COEUR n°0 - Mozzi
//********************************************************************
void setup() {
}

void loop() {
}

//********************************************************************
//     COEUR n°1 - Interface
//********************************************************************
void setup1(void) {
}

void loop1(){
}

```

These return an error "Compilation error: 'StereoOutput' does not name a type".

# Try 1 - Switch to Mono

Switching to Mono mode in mozzi_config.h, and replacing

```c++
return StereoOutput::fromNBit(9, Oscillateur_Scie1.next(), Oscillateur_Scie2.next()) ;
```c++

to

```c++
return MonoOutput::fromNBit(9, Oscillateur_Scie1.next()) ;
```

-> the compilation work. The problem seems linked to the stereo mode.

# Try 2 - Find the disturbing element

I was able to find what make a mess but I don't uderstant why ^^. Moving the function definition of `callback_updateEncodeurRotatif()` after updateAudio and the sketch is compiled witout error :

```c++
//********************************************************************
// 2023 Djevahirdjian Léo
//
// Bug Stereo - Minimal Code
//
//********************************************************************

//********************************************************************
//     DEFINITIONS - Interface
//********************************************************************
// Rotary encoder
# define ROT_PIN0 6
# define ROT_PIN1 7
#include <RotaryEncoder.h>
RotaryEncoder *interface_encodeurRotatif = nullptr ;

//  Definition of the function called when rottary encoder is used.
// THIS MAKE THE COMPILATION FAILS
void callback_updateEncodeurRotatif(){
  interface_encodeurRotatif->tick() ;
}

//********************************************************************
//     DEFINITIONS - Mozzi
//********************************************************************
#include <MozziGuts.h>
#include <AudioOutput.h>
// Template oscillateur
#include <Oscil.h>
// Chargement des tables pour oscillateurs
#include <tables/saw2048_int8.h>
#include <tables/sin2048_int8.h>

Oscil <SAW2048_NUM_CELLS, AUDIO_RATE> Oscillateur_Scie1(SAW2048_DATA) ;
Oscil <SAW2048_NUM_CELLS, AUDIO_RATE> Oscillateur_Scie2(SAW2048_DATA) ;

int CH1_frequence = 440 ;
int CH2_frequence = 220 ;

void updateControl() {
  Oscillateur_Scie1.setFreq(CH1_frequence) ;
  Oscillateur_Scie2.setFreq(CH2_frequence) ;
}

// Fonction de génération de la sortie
AudioOutput_t updateAudio() {
  return StereoOutput::fromNBit(9, Oscillateur_Scie1.next(), Oscillateur_Scie2.next()) ;
}

//********************************************************************
//     COEUR n°0 - Mozzi
//********************************************************************
void setup() {
}

void loop() {
}

//********************************************************************
//     COEUR n°1 - Interface
//********************************************************************
void setup1(void) {
}

void loop1(){
}
```

# Conclusion
I don't have a clue what happen, it's really strange...  I don't know if that is a clue that something wrong is happening with stereo mode... Hope that can help.

Thanks for your good work :-)

Leo

tfry-git, 2023-05-20:
On a general note, it's generally helpful if you try to come up with a "minimal example", i.e. the shortest sketch that still shows the bug. That's both easier for others to follow, and may often help you find a solution, yourself.

Anyway, you are using a sketch that tries to produce stereo output (```return StereoOutput::fromNBit(9, Oscillateur_Scie1.next(), Oscillateur_Scie2.next()) ;```). Probably, however, you forgot to enable stereo in mozzi_config.h (```#define AUDIO_CHANNELS STEREO```).

tfry-git, 2023-05-20:
Ok, first things first: Are you sure you have enabled stereo output in mozzi_config.h?

tfry-git, 2023-05-20:
Sorry for the noise, I see you mention that, specifically.

I guess we still need more diagnostics, though. Please enable display of warnings in Arduino (if you haven't already), and post the full compilation messages. That should probably give some further hints.

tfry-git, 2023-05-20:
And as yet another after-thought: Try moving just the ```#include <MozziGuts.h>``` statement a bit higher up in the sketch. AFAIR, Arduino only looks at the first n lines trying to resolve which libraries to use, so I imagine something strange may be happening, there.