sensorium/Mozzi #194: Compilation errors when including both WaveFolder.h and ResonantFilter.h

techno-womble, 2023-07-25:
Hi,
The following sketch produces multiple compile errors

> #include <MozziGuts.h>
> #include <WaveFolder.h>
> #include <LowPassFilter.h>
> 
> void setup() {
>   // put your setup code here, to run once:
> }
> 
> void loop() {
>   // put your main code here, to run repeatedly:
> }
> 

The errors are repeatable on two PCs and take the following form :-

> _In file included from /home/john/Arduino/libraries/Mozzi/ResonantFilter.h:16:0,
>                  from /home/john/Arduino/libraries/Mozzi/LowPassFilter.h:16,
>                  from /home/john/Desktop/LPFandWF/LPFandWF.ino:3:
> /home/john/Arduino/libraries/Mozzi/IntegerType.h:2:32: error: redefinition of 'struct IntegerType<BYTES>'
>  template<uint8_t BYTES> struct IntegerType {_
>                                 ^~~~~~~~~~~

_

After some playing around I eliminated the compilation errors by adding #ifndef /#define directives to the file IntegerType.h. I must admit, I'm a bit out of my depth here so I thought I'd better log it for someone else to take a look at. 

Many thanks
John

tomcombriat, 2023-07-26:
Hi,
Did not see that coming! Indeed, preproc directives to ensure this file is included only once is the correct solution. I will try to do this today but if you are faster do not hesitate to submit a PR!

Thanks for reporting this!

tomcombriat, 2023-07-26:
Hi again,
@techno-womble could you please give #195 a try and report back (do not have hardware available now).

Thanks!

techno-womble, 2023-07-26:
Tested and works perfectly.
Thanks and keep up the great work!
John

PR #195 (2023-07-26): Fix IntegerType.h
Fix #194
closed 2023-07-26 by techno-womble