sensorium/Mozzi #196: Unable to compile for Atmega328PB with Arduino IDE

M4vrick99, 2023-08-15:
Hello,

I have an issue related to the Atmega328PB.
I'm using the Mozzi library for a small audio project made with the Arduino IDE, I'm programming directly the MCU using the Minicore boards definition and a USBASP programmer. I'm NOT using an Arduino board but only the Atmega328PB on my own board.
There was no issue with my first version using an Atmega328P MCU, but now there si only Atmega328PB available and it seems my code is no more compiling with this version of Atmega328.
I didn't change anything expect switching from 328P to 328PB in the board selection, of course if compiling for 328P it's working but I can't upload the code as the MCU signature is not the good one.

Here are the errors :

`In file included from C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\MozziGuts_impl_AVR.hpp:14:0,

                 from C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\MozziGuts.cpp:35:

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\utility/TimerOne.h: In member function 'void TimerOne::setPwmDuty(char, unsigned int)':

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\utility/TimerOne.h:157:14: error: 'TIMER1_A_PIN' was not declared in this scope

   if (pin == TIMER1_A_PIN)

              ^~~~~~~~~~~~

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\utility/TimerOne.h:157:14: note: suggested alternative: 'TIMER1A'

   if (pin == TIMER1_A_PIN)

              ^~~~~~~~~~~~

              TIMER1A

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\utility/TimerOne.h: In member function 'void TimerOne::pwm(char, unsigned int)':

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\utility/TimerOne.h:175:14: error: 'TIMER1_A_PIN' was not declared in this scope

   if (pin == TIMER1_A_PIN)

              ^~~~~~~~~~~~

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\utility/TimerOne.h:175:14: note: suggested alternative: 'TIMER1A'

   if (pin == TIMER1_A_PIN)

              ^~~~~~~~~~~~

              TIMER1A

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\utility/TimerOne.h: In member function 'void TimerOne::disablePwm(char)':

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\utility/TimerOne.h:208:14: error: 'TIMER1_A_PIN' was not declared in this scope

   if (pin == TIMER1_A_PIN) {

              ^~~~~~~~~~~~

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\utility/TimerOne.h:208:14: note: suggested alternative: 'TIMER1A'

   if (pin == TIMER1_A_PIN) {

              ^~~~~~~~~~~~

              TIMER1A

In file included from C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\MozziGuts.h:224:0,

                 from C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\MozziGuts.cpp:15:

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\MozziGuts_impl_AVR.hpp: In function 'void startAudio()':

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\AudioConfigStandardPlus.h:23:29: error: 'TIMER1_A_PIN' was not declared in this scope

 #define AUDIO_CHANNEL_1_PIN TIMER1_A_PIN // defined in TimerOne/config/known_16bit_timers.h

                             ^

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\MozziGuts_impl_AVR.hpp:197:11: note: in expansion of macro 'AUDIO_CHANNEL_1_PIN'

   pinMode(AUDIO_CHANNEL_1_PIN, OUTPUT); // set pin to output for audio

           ^~~~~~~~~~~~~~~~~~~

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\AudioConfigStandardPlus.h:23:29: note: suggested alternative: 'TIMER1A'

 #define AUDIO_CHANNEL_1_PIN TIMER1_A_PIN // defined in TimerOne/config/known_16bit_timers.h

                             ^

C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master\MozziGuts_impl_AVR.hpp:197:11: note: in expansion of macro 'AUDIO_CHANNEL_1_PIN'

   pinMode(AUDIO_CHANNEL_1_PIN, OUTPUT); // set pin to output for audio

           ^~~~~~~~~~~~~~~~~~~

Using library Mozzi-master at version 1.1.0 in folder: C:\Users\M4vrick\Documents\Arduino\libraries\Mozzi-master 
exit status 1
Error compiling for board ATmega328.`

Hope you can help.

M4vrick99, 2023-08-15:
I think found the issue : the TimerOne library seems not to support the 328PB.

The workaround I used is to add the ATMEGA328PB in the define in known_16bit_timers.h
Dont know if it's the right way to do it but as the pinout is the same it's working. 

In case someone has the same issue before it's fixed.

tfry-git, 2023-11-11:
I suppose you're talking about this line, then: https://github.com/sensorium/Mozzi/blob/cf9e173cd181f2a893f31923aaa0f2ec8e39efbc/config/known_16bit_timers.h#L71C1-L71C66

is ```__AVR_ATmega328PB__``` the proper name of the define to add?

tomcombriat, 2024-03-13:
For reference: https://forum.arduino.cc/t/compilation-error-between-328p-and-328pb-with-mozzi-library/1230299

commit ba7c18a2 (2024-06-12): Add support for ATMEGA328PB

Fixes #196
closed 2024-06-12 by tfry-git