sensorium/Mozzi #197: UNO R4 Support

LightningStalker, 2023-08-23:
I tried compiling one of the sample sketches with the UNO R4 board selected and it seems to think that it is an STM32 pill board. It is looking for headers from the STM32 board files. The R4 is brand new so this is my first time running into something like this.

tomcombriat, 2023-08-23:
Hello! 
Which version of Mozzi are you using? 
The version provided by the Arduino library manager does not support the Uno R4 yet. Only the GitHub version does (until we roll a new Release).

LightningStalker, 2023-08-23:
Ok I was using the one from the librarly manager. I will try the latest build.
I was also poking around and noticed a definition of BSP_MCU_GROUP_RA4M1 but I see you used ARDUINO_FSP to detect the core. I'm not sure what FSP and BSP are but we are living in an age of new things.

tomcombriat, 2023-08-23:
If I remember correctly, FSP is an hardware abstraction layer.
This port is fairly recent, do not hesitate to report here if it works correctly (or not), which will be one step further to make a new release so that the Arduino library manager has access to it!

LightningStalker, 2023-08-23:
Ok thanks for clarification.
So right now I'm running the sample playing example on a Minima and the output doesn't sound quite right. The burroughs sample (forgive me for not knowing what that is or what it's supposed to sound like) sounds like a little click and the raven sample is inaudible. It could be something to do with the sample rate but I'm not quite sure.

sensorium, 2023-08-24:
Hi,
I just tested the example 08.Samples>Sample on a pro micro and it's working here.  

Also, audio output from the examples, to let you know what they should be doing, is here: https://sensorium.github.io/Mozzi/examples/

LightningStalker, 2023-08-24:
Thanks sensorium, it doesn't sound anything like that. All I'm getting from the Sample sketch is a clicking sound.

tomcombriat, 2023-08-26:
Hi,
Okay, that's a good start already! Thanks for reporting this.
I will have a look at it once I get back close to a computer (a couple of weeks probably). Maybe someone else will sold it in the meantime.

I am actually wondering if this example would not have been broken because of a previous change and not necessarily on this port only, I've never tested it...

tomcombriat, 2023-09-09:
Hi,
Sorry for the delay, but I am finally back behind my breadboard.
I tested the `Sample` sketch (and others in the same folder) on both R4 Minima and R4 Wifi and it works with no problem here.

As it seems that you are not having the expected output a few questions arise:

- Are you running the example sketch with no modification?
- Did you make any change in the config files of Mozzi?
- How are you outputting the sound (connections)?

LightningStalker, 2023-09-10:
In regards to the sample playing sketch that was not working,

- No modifications to the sketch were made
- No changes to the Mozzi config files were made
- I have an LM386 board connected to pin A0 (DAC) (the LM386 is connected to a speaker)

All of this is the exact same configuration as for the FMsynth example which is working.
Now I think there were some updates to the core which might make a difference.
The IDE was also updated which in theory shouldn't make any difference.
I will try running it again but not now it's late for me here.

tomcombriat, 2023-09-10:
Okay, thanks! Indeed, as the FMSynth does work the circuit is very likely to be correct, you probably said it before but I overread it, sorry! Normally updates on the core should not impact as the development was initially done on a early release version of the core but who knows?

I am trying to finish this port now (which async adc) in order to push a new release. Hopefully this might bring more users of the R4 and we will get more feedback…

tomcombriat, 2025-01-14:
@LightningStalker Is it solved?