sensorium/Mozzi #208: Please Change Mozzi's License

eclab, 2023-11-09:
Mozzi's choice of open source license is poor and it should be relicensed under a better license.

Mozzi is licensed under CC-NC-SA-International, a license EXPRESSLY not meant for code, but rather for text, documentation, pictures, sound, video, and other media.  Indeed, Creative Commons itself strongly warns against its use for code:   

https://creativecommons.org/faq/#can-i-apply-a-creative-commons-license-to-software

Source code is different from other media in important ways.  First, it is the only medium which can be **both** copyrighted **and** patented, leading to a rat's nest of legal entanglements, and good open source licenses (Apache, GPLv3) take this into account.  CC, not being meant for code, does not even consider the issue.  Second, code is designed to be integrated with other code, meaning that their licenses must be compatible.  Creative Commons licenses cannot be integrated with other proper software licenses at all, including both Apache and GPL, not to mention a host of others.  As Creative Commons has pointed out, Mozzi projects which use GPL or Apache are presently in license violation.  Finally, code falls under many international legal issues that do not befall other kinds of media, and CC does not deal with them.

To make matters worse, Mozzi is licensed under a viral license with no library release, meaning that to use its library we must relicense our own code under the same broken license.  

Please relicense Mozzi under an open source license.  My suggestion would be Apache 2.0, which makes Mozzi available to the widest audience.  But if you insist on a viral license, please at least relicense it under GPLv3.

sensorium, 2023-11-11:
Licensing is painful and I could do with some help.  I wrote Mozzi for my own totally unsuccessful art projects and I thought it could be useful for others.  It's not a grand project.  Please don't beat me around for it.

I found CC-NC-SA when I was hoping to let people do their stuff non-commercially, and get in touch if they want to use Mozzi commercially.  Is there a simple way to do this?

tfry-git, 2023-11-11:
@sensorium I think @tomcombriat and I can help with that, but we definitely need your ok for any path we take.

To sum up the basic options really **roughly**:

- Apache 2.0 means you can use Mozzi code pretty much any way you like. https://www.tldrlegal.com/license/apache-license-2-0-apache-2-0
- LGPL means you can use Mozzi as is in any way you like, *but* if you make changes to the library itself, these need to be LGPL, again. https://www.tldrlegal.com/license/gnu-lesser-general-public-license-v3-lgpl-3
- GPL means you can use Mozzi in any way you like, including commercially, but you need to offer the source of your project under GPL. https://www.tldrlegal.com/license/gnu-general-public-license-v3-gpl-3
- (Both GPL and LGPL come in v2 and v3 variants. I'm not sure, how relevant v2 still is these days, but when going with either of these, I suggest using the specification "version 2 or later", to be on the safe side.
- None of the above option precludes that you, **the copyright holder**, grant additional licenses to whomever under whichever terms.

I'm fine with all three of the above with a small tendency towards LGPL. If you (and @tomcombriat) are ok with one of these, I'll be happy to come up with a PR (eventually).

tfry-git, 2023-11-11:
Some more detail, addressing commercial use, specifically: 
> I found CC-NC-SA when I was hoping to let people do their stuff non-commercially, and get in touch if they want to use Mozzi commercially. Is there a simple way to do this?

Again, none of the above really precludes commercial use. Apache 2.0 would mostly mean commercial use is always allowed. For GPL commercial use would be somewhat complicated in that you are obliged to offer the sources in GPL. For LGPL, that depends.

But again, nothing precludes you as copyright holder from allowing further use, and we could put a statement to that effect in the Readme.

eclab, 2023-11-11:
Nobody beatin' nobody up here.  Early on I used lots of amateur licenses too.

Here are your options in decreasing order of restrictions:

0. Restrictions on commercial use are against the notion of open source.  More importantly, an NC license is not compatible with Arduino's LGPL, so licensing as NC legally precludes people from using your code with Arduino's code!  That ain't good.

1. If your goal is to get a piece of the pie when Mozzi is used commercially, there is really only one option: a dual private and GPL license.  Basically you offer Mozzi under GPL, but if a company wants to use Mozzi without having to release their code as GPL, they have the option to negotiate with you for a nontransferable, nonexclusive private license instead.

It's tricky though.  If people make contributions to Mozzi, they would be under GPL and you *cannot* offer these as part of the private license.  So all contributions or revisions by people other than yourself would have to be in a contrib directory which can be separated.  It's not great but there you have it.

2. If your goal is to make sure that companies do not use your stuff without contributing back when *selling products*, then the obvious choice is to license everything as GPL 3.0.

3. If you're fine with companies using your stuff and not contributing back when selling products, but you want companies that *change* or *improve* your stuff in their products to have to contribute back, then the obvious choice is LGPL.

4. If you're fine with anyone using your stuff for any reason without contributing back, the clear choice is Apache 2.0.

As a rule I license everything I make -- a lot of stuff mind you -- as Apache 2.0, as I want my code available to as wide an audience of people as possible.  But if you're concerned about commercial exploitation, I think your best option is probably #2: GPL.  You could do #1 but it will be real hassle.

Don't think of Mozzi as an opportunity to make money... think of it as an opportunity to become famous!  And a way to make a contribution!

Sean

tomcombriat, 2023-11-11:
As @tfry-git said, I am also ready to help in the effort of re-licensing (maybe a well defined `sed`? would do the trick). As per the license, I do not have any strong opinion but I am leaning a bit more for LGPL. But I think this should be your choice in the end @sensorium and will be fine whatever it is. 

>  It's not a grand project.

I think it is a great project and you should not under estimate your work @sensorium ;).

eclab, 2023-11-20:
Was hoping to contribute to Mozzi soon!  Any chance this could be bumped?  I'd be glad to provide code with revised license information if a decision was reached with regard to what license to use.  I myself would do Apache 2.  But given @sensorium's (reasonable!) opinions regarding commercialism, I would suggest GPL 3.

sensorium, 2023-11-23:
... still thinking ...

sensorium, 2023-11-23:
@tfry-git and @tomcombriat, why do you both lean towards LGPL?

tfry-git, 2023-11-23:
LGPL vs Apache / BSD, other permissive license: I think the concept of "share alike" is not too much to ask for in general. Seeing, e.g. how Apple builds on BSD code and then shows massive disrespect for the open source community is something that really makes me angry. (Substitute with whatever company triggers your rage.) Now, I don't actually foresee this happening with Mozzi, but it makes me think that some concept of "share alike" should be the "natural" norm to live by in the open source community.

LGPL vs GPL: GPL takes the above thought furthest, but at the same time, I acknowledge that in some cases GPL may be a bit restrictive, when trying to combine work from several different projects. Then again, we're talking about embedded systems, here, and thus it seems reasonable to expect that any project using Mozzi will be using Mozzi as a relatively central feature, so in turn it seems reasonable enough, if it enforces a particular license. (In contrast, picture e.g. a large office suit that might use a sound synthesis library only to generate some exchangeable notification sound in one of its component apps).

So in effect, I do lean towards LGPL, but it's a close call between GPL and LGPL to me. Apache is a mere third choice (but still quite acceptable) to me.

tomcombriat, 2023-11-25:
I basically have the same thoughts than @tfry-git. LGPL might attract a bit more users than the GPL which in turn, might bring more collaborations and additions of features to Mozzi.

closed 2024-06-09 by tfry-git