sensorium/Mozzi #209: Using Mozzi and I2C library for LCD

bbassac, 2023-11-11:
I try to use Mozzi (1.1.1) and a library for LCD (tried with LiquidCrystal_I2C and hd44780)
Used alone in a blank project without Mozzi, works fine.

Once used in the program, i've got a conflit : 
`:\Temp\arduino\sketches\D6CCD9D2BA2F2BBFEB8F090DBBA13BEB\libraries\Mozzi\twi_nonblock.cpp.o (symbol from plugin): In function `initialize_twi_nonblock()':
(.text+0x0): multiple definition of `__vector_39'
G:\Temp\arduino\sketches\D6CCD9D2BA2F2BBFEB8F090DBBA13BEB\libraries\Wire\utility\twi.c.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status`

I think it's related with Wire.h and timers, but i don't know where the problem is from.
Common point is Mozzi

sensorium, 2023-11-11:
Hi there,
try removing the files twi_nonblock.h and twi_nonblock.cpp from the Mozzi library folder.

If it compiles after that, you might get audio glitches with the LCD.  Maybe it's worth asking on the discussion forum if anyone's used that device.  Otherwise, you're looking at writing some non-blocking code yourself, or working around the glitches (eg. pause the audio while you change the display, or your own more clever idea).  twi_nonblock is there as a starting point and example to help with writing your own nonblocking code for the device.

tfry-git, 2023-11-13:
Look up the pins in config/known_16bit_timers.h . This will be TIMER1_A_PIN and TIMER1_B_PIN

However, it may also be a good idea to try STANDARD_PLUS mode just for testing purposes.

bbassac, 2023-11-15:
My output circuit seems good (checked resistor and capacitor values), bu
got some ugly noise (tried with the example of simple midi synth and a sin
waveform,
I can hear that there is à sin in the sound, but is covered with noise .

Ca you think i can plus a DAC on it  ? (it's a i2c one, and i fear having
same problems thant using a LCD with the  initialize_twi_nonblock()'....

tfry-git, 2023-11-15:
Just stumbled across this in your earlier post:
> HIFI_STANDARD_PLUS
This mode does not exist. There's HIFI, and there's STANDARD_PLUS, but never both at once. Could you post your mozzi_config.h, somehwere?

> I can hear that there is à sin in the sound, but is covered with noise .

Is that with the LCD, or without?

1) If it is without the LCD: Something is wrong. Try if "STANDARD_PLUS" works. If that still doesn't work any better, something is wrong in your sketch.

2) If it is with the LCD: Try without any LCD code, just to make sure. If you can thus pin down the problem to the LCD, using an external DAC is not likely to help, however. In this case, try, whether it helps any to keep updates to the LCD to a minimum (e.g. call LCD functions only on every 100th run of updateControl()).

3) About connecting a DAC: The key problem with I2S communication, is that is is often implemented in a blocking manner. I.e. the CPU sits idle while waiting for a reply. Meanwhile the audio output buffer is depleting... Are you planning on using a library for connecting to the DAC? You may want to check, what's documented, there.

tomcombriat, 2023-11-15:
> Ca you think i can plus a DAC on it  ? (it's a i2c one, and i fear having same problems thant using a LCD with the  initialize_twi_nonblock()'....

Note that I2C (not I2S) DACs are usually not a good idea for audio (at least in my experience): the I2C protocol is quite slow so it is difficult to attain the desire rate.

As @tfry-git , I would suggest taking an incremental approach, from bottom to top:
 - without LCD, in STANDARD_PLUS, can you output a clean sound?
 - if yes, try adding the LCD, with minimal writes to it at first
 - if that works, you can switch to HiFI,

bbassac, 2023-11-15:
Hi, i made a mistake with writing this post.
I have this in my mozzi_config.h :

#define AUDIO_MODE STANDARD_PLUS
Are we ok that this mode makes a beter sound ?
I made this circuit, and plugged on my Arduino Mega between pins 11 and 12
//
#elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
  #define TIMER1_A_PIN   11
  #define TIMER1_B_PIN   12

Does the 11 is on the 3.9 K Ohms side, and the 12 on the 499k ohm side  ?
I thought to the DAC (a MPC4725) because lot of people says that it's the
better sound, bu don't know if mozzi can handle it without any external
librairy....

tfry-git, 2023-11-16:
STANDARD_PLUS uses just one pin for output. The circuit you are using is for a different output option.

However, _do_ start with STANDARD_PLUS: Audio output is simply to pin 11 (on the Arduino Mega), then, no additional components required (not for basic testing, anyway). Quality will not be super great, but you will definitely get a signal with a little noise, not the other way around. Does this work? Ok, _then_ you can proceed to https://sensorium.github.io/Mozzi/learn/output/ for increasing the output quality (if needed).

bbassac, 2023-11-20:
I'll try the standard_plus way for beginning.
This is the expected plan : 
[image]

tomcombriat, 2023-11-20:
Looks good from here!

Maybe irrelevant, depending on how it is actually done but if I am correct, when no Jack is inserted this produces a hard short circuit between A11 and GND. I'm never sure of the protection of these pins, but I would not try it out ;)

poetaster, 2024-03-27:
I had a question about a dynamic way to stop loading two_ Most of my sketches with mozzi don't use a display but one I'm working on does (using the adafruit SSD1306 libary). Could one modify twi_nonblock to load with a define?

poetaster, 2024-03-27:
Just for background, I'm trying to get my bytes noise sketches from aduboy/midiboy ported to my rampart (which is designed for mozzi stuff) https://github.com/poetaster/rampart Midiboy is what I wrote them on https://github.com/BlokasLabs/Midiboy/tree/master/src but the code is mostly from Arduboy (ie. 328). Looking at that, the display drive is not I2C but the interrupts level audio and drawing routines work flawlessly. Probably because the games people always make sure it works :)

poetaster, 2024-04-05:
And, I wanted to add that with Mozzi version 2, PWM output ( custom pin) if I use the second cpu I can use the adafruit libraries and I2C bus for display and muxed input on the RP2040 (a pico). To reduce noise, don't forget:
```
  pinMode(23, OUTPUT); // thi is to switch to PWM for power to avoid ripple noise
  digitalWrite(23, HIGH);
```
which keeps the pico from dropping into PFM power saving mode and forces PWM mode. That's quiet. Running from a 3.7 volt lithium with the display active and a simple RC output, it's really nice! I have to see what, if anything is possible with the beloved nano :)

tfry-git, 2024-04-05:
Nice find. We should probably either do this from startMozzi(), or document it as a suggestion.