sensorium/Mozzi #213: UNO R4 port broken

tomcombriat, 2023-11-28:
The UNO R4 port seems to be broken when using the UNO R4 core from Arduino and not the Renesas core (which was the one given for early adopters…).

The only annoying thing is that SPI seems to be wayyyyyy faster on the UNO R4 core compared to Renesas (even a single DAC with Mozzi does not work well).

I'll try to commit myself to see how to do the port for this core. Wonder if both the Renesas and R4 port should both stay or not… The way the Renesas DAC could be hooked up to the Mozzi buffer was quite neat, hopefully this is not that different.

@tfry-git do you prefer that I start porting from the `config/rework` or from `master`?

tfry-git, 2023-11-29:
[I'll be a bit slow for some more time]

> @tfry-git do you prefer that I start porting from the config/rework or from master?

Both has its set of drawbacks. I suggest to start porting in config_rework, and then to evaluate how much trouble it will be to backport the changes.

tomcombriat, 2023-11-29:
> [I'll be a bit slow for some more time]

Me too actually, and I never post thing to press, but just in case I forget ;).

> . I suggest to start porting in config_rework, and then to evaluate how much trouble it will be to backport the changes.

In the end, I did *two* PR, one of config_rework, and one on master. As the changes are quite small, this does not seem to conflict with anything. Merging these two would allow the config_rework to stay synced with master, easying the futur big merge. Also, that makes the port for R4 actually working (with a release, available on Arduino IDE).

PR #214 (2023-12-09): Fix of mozziAnalogRead for Renesas
Fixes #213
closed 2023-12-09 by tomcombriat