sensorium/Mozzi #232: RFC: Analog input scaling - what's the best approach

tfry-git, 2024-01-29:
Since some years, updateAudio() can be platform independent in so far as scaling to the required audio resolution can be done, automatically.

Where we still have differences is analog (audio or non-audio) input. mozziAnalogRead(), and getAudioInput() return values in a platform specific range. It should be possible to address this quite easily, but adding another configuration define like:

```
#define MOZZI_ANALOG_READ_RESOLUTION 12
```

All ports will then either have to make sure they perform reads at that resolution, _or_ they can set some internal define telling what resolution analog reads actually have. Then mozziAnalogRead() and getAudioInput() can trivially return values that have been shifted, accordingly.

So far so good. What I'd like to pick your brains about: What should be the default behavior, when users do not set MOZZI_ANALOG_READ_RESOLUTION? Essentially, I think the options are:

  - Continue to return reads in platform specific resolution -> what we have now.
  - Return reads at 10 bits resolution -> the default on classic AVR. Breaks some sketches (those written specifically for other platforms), fixes some sketches (those written for AVR, but used on other platforms)
  - Return reads at 16 bits resolution -> enough room on all platforms, without exceeding uint16_t range. Breaks existing sketches, however.

Thoughts?

tomcombriat, 2024-01-29:
I thought about that some time ago, and my ideas converged on the opposite approach: I came to the idea that every port should also give, as an accessible parameter say `NATIVE_ANALOG_READ_RESOLUTION`.

With that, sketches can be made platform independent for instance by doing:
```
uint8_t midi_note = MOZZI_SHIFTR(mozziAnalogRead(pin), NATIVE_ANALOG_READ_RESOLUTION - 8); // returns a 8bits value, whatever the input range
```
(reusing the `MOZZI_SHIFTR` from FixMath ;).

I did not thought it completely through, but that might also be an approach to consider.

tomcombriat, 2024-01-29:
Otherwise, I would be in favor of returning to 16 bits resolution: 10 bits do require a 16 bits container anyway, better not to be blocked in the future and provide the best resolution, by default, for platforms that can provide it.

tfry-git, 2024-01-30:
> Also, we are breaking a lot of things for Mozzi2.0 already…

I'm still sort of optimistic, that most sketches will continue to work, even if perhaps with warnings.

Anyway, building on your feedback, my next suggestion would be:

If no desired analog resolution is configured, ```mozziAnalogRead()``` continues to return the platform specific resolution, but with a warning. If something is configured, use that, without a warning.

In addition, create a new function ```mozziAnalogRead16()```, which always returns the result shifted to 16 bits. (Possibly also ```mozziAnalogRead<byte resolution>()```, but that may be overkill).

(Rationale: We're only dealing with two functions, here, ```mozziAnalogRead()``` and ```getAudioInput()```, so integrating the wrapper into the function will me more comfortable, without adding much API bloat.)

closed 2024-06-09 by tfry-git