sensorium/Mozzi #246: GP0 io issues on RP2040

poetaster, 2024-04-05:
GP0 seems  to be contentious? I've maxed out all pins and have custom assignments:
```
#define AUDIO_CHANNEL_1_PIN 22
#define MOZZI_AUDIO_PIN_1 22
``` 
which work, but GP0 appears to be pulled up high at some point? 

I noted the default output is on 0 so thought maybe ...

If I put startMozzi before my pinMode setups, GP0 is initially usable but then just goes high after some 10s of seconds. Could be a strange bug in my sketch (which is a bit messy),  it's this https://github.com/poetaster/scarp/tree/main/firmware/MozziFM2 sketch.

poetaster, 2024-04-05:
I just tried adding extra debounce logic, and I don't get it. Sometimes it works for a minute and then the pin goes high. Other sketches and programs not made with arduino and mozzi)  don't exhibit this behaviour. I don't think it's hardware. Still, could be something in the sketch, but much of it is copied from a sketch (without mozzi) which just works. in particular the button/led handling.

poetaster, 2024-04-05:
Sorry for so much signal(hope :): setup looks like:
```
  envelope.setADLevels(255, 200);
  envelope.setTimes(50, 200, 1000, 200); // 10000 is so the note will sustain 10 seconds unless a noteOff comes

  startMozzi();
  
  displaySplash();
  
  pinMode(BUTTON0, INPUT_PULLUP);
  pinMode(BUTTON1, INPUT_PULLUP);
  pinMode(BUTTON2, INPUT_PULLUP);
  pinMode(BUTTON3, INPUT_PULLUP);
  pinMode(BUTTON4, INPUT_PULLUP);
  pinMode(BUTTON5, INPUT_PULLUP);
  pinMode(BUTTON6, INPUT_PULLUP);
  pinMode(BUTTON7, INPUT_PULLUP);
  pinMode(SHIFTBUTTON, INPUT_PULLUP);

  pinMode(26, INPUT);
  pinMode(27, INPUT);

  pinMode(LED0, OUTPUT);
  pinMode(LED1, OUTPUT);
  pinMode(LED2, OUTPUT);
  pinMode(LED3, OUTPUT);
  pinMode(LED4, OUTPUT);
  pinMode(LED5, OUTPUT);
  pinMode(LED6, OUTPUT);
  pinMode(LED7, OUTPUT);

  pinMode(LED, OUTPUT);
  
  pinMode(23, OUTPUT); // thi is to switch to PWM for power to avoid ripple noise
  digitalWrite(23, HIGH);
```

I just tried NOT playing button0 and after about 40 seconds, it 'just went high'. 

Jeez I hope this is not a bug on my side :)

tfry-git, 2024-04-05:
Suspicious line: https://github.com/poetaster/scarp/blob/3fdb3176e36af1025a423b88fec484796f1bccb7/firmware/MozziFM2/MozziFM2.ino#L464

I believe you want ```< 8```, and ```<= 8```, here.

poetaster, 2024-04-06:
> Suspicious line: https://github.com/poetaster/scarp/blob/3fdb3176e36af1025a423b88fec484796f1bccb7/firmware/MozziFM2/MozziFM2.ino#L464
> 
> I believe you want `< 8`, not `<= 8`, here.

Nope, there are 9 buttons. 
```
#define NUM_BUTTONS 9 
int buttons[NUM_BUTTONS] = {BUTTON0, BUTTON1, BUTTON2, BUTTON3, BUTTON4, BUTTON5, BUTTON6, BUTTON7, SHIFT}; 
```
 I could write < 9. Did, just for the sake of it, try that :) That's not it. That pin is (the button on GP0) being put high, I believe, by mozzi.

That's the same loop (well, same io) as in https://github.com/poetaster/scarp/blob/main/firmware/Pikobeats/Pikobeats.ino

In the latter, the 0th led/button don't fire 'randomly'. I thought maybe it's the debounce logic but that's the same (currently) as in Pikobeats.

tfry-git, 2024-04-06:
> Nope, there are 9 buttons.

But only 8 leds?

poetaster, 2024-04-06:
Yes, only 8 leds. That's true. That ```digitalWrite(led[i] , HIGH);``` in the loop could be an issue IF you actually use that button :) 

I'm just toying around and think it could have something to do with the encoder which, when GP0 goes high, is no longer responding UNLESS one of the other buttons is pressed. Hmmm. Still, weird that it works elsewhere.

closed 2024-04-06 by poetaster