sensorium/Mozzi #250: A question, not an issue per se:  lgt8f328p

poetaster, 2024-04-11:
I'm not sure if you guys are aware of the LGT8F328P – LQFP32 Boards ? All of my Nano specific Arduino sketches (including custom timer stuff) 'just work'. Including Mozzi. A good review: https://wolles-elektronikkiste.de/en/lgt8f328p-lqfp32-boards#comment-27646

In comparison with the Nano, they have an extra 16 bit timer (all timers up to 64MHZ) and an actual 8bit DAC  (on pin D4) that you can just analogwrite to. The internal osc is 32MHZ, 12 channels ADC up to 12 bit and with gain . They can be purchased for less than you pay for the atmel counterpart.

I'm moving my rampart projects to this chip. But I haven't done anything interesting yet.

tfry-git, 2024-04-11:
Yes, I stumbled across those not too long ago, and yes, Mozzi seems to just work on them. Quite neat, indeed. So I've added them to the table of supported boards for Mozzi 2.0 (https://tfry-git.github.io/Mozzi/), even though they're simply using the AVR codepath, so far.

tfry-git, 2024-06-12:
Closing, as this is now documented supported.

Eventually, if there is enough interest, we may want to add specific support for the DAC, of course.

closed 2024-06-12 by tfry-git