sensorium/Mozzi #258: Teensy (3.6 and 4.1) CI jobs are failing

tfry-git, 2024-06-13:
This does not appear to be our fault, currently. E.g. https://github.com/sensorium/Mozzi/actions/runs/9488741406/job/26164634199#step:4:199

I tried re-running the action for a previously successful commit (https://github.com/sensorium/Mozzi/commit/6b4ba48cf29258d7c64cf33886b758738c8b528f), and it failed, there, too. 

Error message: 
>   Opening Teensy Loader...
>  Unable find Teensy Loader.  (p)  Is the Teensy Loader application running?
>  quit

We may possibly have to adjust something in the action setup, though.

tfry-git, 2024-06-21:
Updates:
 - downgrading the failing compile-sketches action to v1.1.0 or even 1.0.0 does not help.
 - I posted about this at the teensy forum, but got no reply https://forum.pjrc.com/index.php?threads/teensy-boards-starting-to-fail-in-github-actions-arduino-compile-sketches.75247/

tomcombriat, 2024-08-01:
Apparently got solved by the latest CLI? https://github.com/arduino/arduino-cli/issues/2658

tfry-git, 2024-08-01:
Yes, already removed our workaround (https://github.com/sensorium/Mozzi/commit/acda05409538d4d5cd48cd7bd80caf503465997a) but forgot to close the issue.

closed 2024-08-01 by tfry-git