sensorium/Mozzi #261: ESP32 21:58:07.296 -> E (200) ADC: CONFLICT! driver_ng is not allowed to be used with the legacy driver

kicetwigs, 2024-06-20:
can't upload any example sketch to ESP32 board, sketch always reboots
mozzi 2.0. 

22:00:41.562 -> E (200) ADC: CONFLICT! driver_ng is not allowed to be used with the legacy driver
22:00:41.562 -> 
22:00:41.562 -> abort() was called at PC 0x400dd62b on core 0
22:00:41.562 -> 
22:00:41.562 -> 
22:00:41.562 -> Backtrace: 0x400828e1:0x3ffe3b40 0x40088b51:0x3ffe3b60 0x4008db2d:0x3ffe3b80 0x400dd62b:0x3ffe3c00 0x400dfb2a:0x3ffe3c20 0x40082c3e:0x3ffe3c50 0x400794fe:0x3ffe3c90 |<-CORRUPTED
22:00:41.562 -> 
22:00:41.562 -> 
22:00:41.562 -> 
22:00:41.562 -> 
22:00:41.562 -> ELF file SHA256: ab3a29cc9dc586b7
22:00:41.562 -> 
22:00:41.562 -> Rebooting...
22:00:41.594 -> ets Jul 29 2019 12:21:46
22:00:41.594 -> 
22:00:41.594 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:00:41.594 -> configsip: 0, SPIWP:0xee
22:00:41.594 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:00:41.594 -> mode:DIO, clock div:1
22:00:41.594 -> load:0x3fff0030,len:1448
22:00:41.594 -> load:0x40078000,len:14844
22:00:41.594 -> ho 0 tail 12 room 4
22:00:41.594 -> load:0x40080400,len:4
22:00:41.594 -> load:0x40080404,len:3356
22:00:41.594 -> entry 0x4008059c
22:00:41.791 -> E (200) ADC: CONFLICT! driver_ng is not allowed to be used with the legacy driver
22:00:41.791 -> 
22:00:41.791 -> abort() was called at PC 0x400dd62b on core 0
22:00:41.791 -> 
22:00:41.791 -> 
22:00:41.791 -> Backtrace: 0x400828e1:0x3ffe3b40 0x40088b51:0x3ffe3b60 0x4008db2d:0x3ffe3b80 0x400dd62b:0x3ffe3c00 0x400dfb2a:0x3ffe3c20 0x40082c3e:0x3ffe3c50 0x400794fe:0x3ffe3c90 |<-CORRUPTED
22:00:41.791 -> 
22:00:41.791 ->

tomcombriat, 2024-06-20:
Ouch… I am really not familiar with the ESP32, but as a first test to narrow down things, does it work with Mozzi1?

tfry-git, 2024-06-21:
Is that a Mozzi example sketch, or something else?

I suppose it's probably related to the fact that the i2s driver was reworked in the ESP32 IDF. Mozzi is still using the old driver, while apparently, something in your sketch is trying to use the new version, at the same time.

We'll probably have to port to the newer driver, sooner or later, but of course that'll mean retesting everything i2s related.

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2s.html

kicetwigs, 2024-06-21:
Hi guys 
Thanks for your reply
I was testing mozzi official examples from the library (basic sine, fm synth,...)
Using latest 1.x mozzi and 2.0

thomasfredericks, 2025-01-09:
What is the latest working version of ESP IDF that was tested working with ESP32? I am running into the same issue.

MatteoHMarangoni, 2025-01-10:
I've been running into this same problem. 

I checked in the Arduino IDE and the last compatible version of the ESP32 definition in the board manager is 2.0.17.

tomcombriat, 2025-01-11:
Very good to know, thanks! 

If you fancy, some help can be needed on #301 ;)

closed 2025-02-24 by tomcombriat