sensorium/Mozzi #262: Mozzi not working on STM32

justinjools22, 2024-06-28:
I tested Mozzi code on Arduino Nano and it works but when I upload to STM32F103C8T6 blue it doesn't do anything. I have tested non-library wavetable sine code on STM32 which works with PWM on PA9. 

I believe it's some software thing. The basic pwm Mozzi examples should just work as on Arduino Nano.

tfry-git, 2024-06-28:
Hi!

Which version of Mozzi, and which STM32-core are you using? (Mozzi supports two different STM32 cores, and the output will be to different pins).

tomcombriat, 2024-06-28:
Hi,
Have you tried also the basic examples (not with an external Dac as you described on the forum)?

justinjools22, 2024-06-28:
I'm using STM32F103C8T6 blue and the latest Mozzi 2.0.0 download in Arduino
library. My destination platform, what I want to use is  STM32F411CEU6
black though.

justinjools22, 2024-06-28:
Yes, I tried the Mozzi basic sine wave pwm which works on Arduino Nano but
not on STM32F103C8T6 blue. If I try the Arduino library basic sine example,
this works on STM32. I can't get any Mozzi code to work on STM32.

tfry-git, 2024-06-28:
> I'm using STM32F103C8T6 blue and the latest Mozzi 2.0.0 download in Arduino library. My destination platform, what I want to use is STM32F411CEU6 black though.

What I meant is the software package you installed from the Board Manager (yes, it's confusing, that there are several). Anyway, perhaps the issue is as simple as looking at the wrong pin? Mono output would be expected on PA8 and PB8, by default, while you mentioned PA9?

justinjools22, 2024-06-28:
I downloaded package from
https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

I have pwm output on PA8 (D9 is Arduino Nano).

justinjools22, 2024-06-28:
I'm also getting this error on STM32, whereas it works on Arduino Nano:

In file included from
c:\Users\m\Documents\Arduino\libraries\Mozzi/internal/config_checks_generic.h:38:0,
                 from
c:\Users\m\Documents\Arduino\libraries\Mozzi/MozziGuts.h:31,
                 from
c:\Users\m\Documents\Arduino\libraries\Mozzi/Mozzi.h:33,
                 from
C:\Users\m\AppData\Local\Temp\.arduinoIDE-unsaved2024528-10508-l4t72n.z93ph\Sinewave\Sinewave.ino:22:
c:\Users\m\Documents\Arduino\libraries\Mozzi/internal/mozzi_macros.h:93:47:
error: missing binary operator before token "("
 #elif defined(__GNUC__) && __has_cpp_attribute(deprecated)
                                               ^
In file included from
c:\Users\m\Documents\Arduino\libraries\Mozzi/internal/MozziGuts_impl_STM32duino.hpp:40:0,
                 from
c:\Users\m\Documents\Arduino\libraries\Mozzi/internal/MozziGuts.hpp:49,
                 from
c:\Users\m\Documents\Arduino\libraries\Mozzi/MozziGuts.h:205,
                 from
c:\Users\m\Documents\Arduino\libraries\Mozzi/Mozzi.h:33,
                 from
C:\Users\m\AppData\Local\Temp\.arduinoIDE-unsaved2024528-10508-l4t72n.z93ph\Sinewave\Sinewave.ino:22:
c:\Users\m\Documents\Arduino\libraries\Mozzi/internal/MozziGuts_impl_STM32duino_analog.hpp:69:2:
error: #error "ADC sampling time could not be defined for internal
channels!"
 #error "ADC sampling time could not be defined for internal channels!"

justinjools22, 2024-06-28:
Finally got it working.

Selecting Generic STM32F1 series, Board part number BluePill STM32F103C6
Connected to PA8

DAC MCP4822 now works with code example:
FMsynth_MCP4921_mono_12bits
#define SS_PIN PB1

It sounds so much clearer and hi-fi that 8 bit pwm on Arduino Nano :)

Thanks for listening. If often just requires running it by someone and
reviewing it.

closed 2024-07-01 by tomcombriat