sensorium/Mozzi #263: MIDI port disappearing when adding Mozzi to RP2040

leobel96, 2024-06-30:
Hello, 
I have this very simple script on a Raspberry Pico W that is working as expected **without Mozzi** (I'm able to send MIDI packets and to see the messages being printed):

```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

#include "user.h"
#include "main.h"

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void setup() {
  // Manual begin() is required on core without built-in support e.g. mbed rp2040
  if (!TinyUSBDevice.isInitialized()) {
    TinyUSBDevice.begin(0);
  }

  pinMode(LED_BUILTIN, OUTPUT);

  usb_midi.setStringDescriptor("TinyUSB MIDI");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);

  MIDI.setHandleNoteOn(handleNoteOn);
  MIDI.setHandleNoteOff(handleNoteOff);
  MIDI.setHandleControlChange(handleControlChange);

  Serial.begin(115200);
}

void loop() {
  #ifdef TINYUSB_NEED_POLLING_TASK
  // Manual call tud_task since it isn't called by Core's background
  TinyUSBDevice.task();
  #endif

  // not enumerated()/mounted() yet: nothing to do
  if (!TinyUSBDevice.mounted()) {
    return;
  }
  // read any new MIDI messages
  MIDI.read();
}

void handleNoteOn(byte channel, byte pitch, byte velocity) {
  // Log when a note is pressed.
  Serial.print("Note on: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleNoteOff(byte channel, byte pitch, byte velocity) {
  // Log when a note is released.
  Serial.print("Note off: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleControlChange(byte channel, byte pitch, byte velocity) {
  // Log when a note is released.
  Serial.print("Control change: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}
```

When I add Mozzi, though, the MIDI port doesn't get opened at all and no message is printed:
```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

#define MOZZI_CONTROL_RATE 128 // Hz, powers of 2 are most reliable
#include <Mozzi.h>

#include "user.h"
#include "main.h"

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void setup() {
  // Manual begin() is required on core without built-in support e.g. mbed rp2040
  if (!TinyUSBDevice.isInitialized()) {
    TinyUSBDevice.begin(0);
  }

  pinMode(LED_BUILTIN, OUTPUT);

  usb_midi.setStringDescriptor("TinyUSB MIDI");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);

  MIDI.setHandleNoteOn(handleNoteOn);
  MIDI.setHandleNoteOff(handleNoteOff);
  MIDI.setHandleControlChange(handleControlChange);

  Serial.begin(115200);

  startMozzi();
}

void updateControl() {
  #ifdef TINYUSB_NEED_POLLING_TASK
  // Manual call tud_task since it isn't called by Core's background
  TinyUSBDevice.task();
  #endif

  // not enumerated()/mounted() yet: nothing to do
  if (!TinyUSBDevice.mounted()) {
    return;
  }
  // read any new MIDI messages
  MIDI.read();
}

void handleNoteOn(byte channel, byte pitch, byte velocity) {
  // Log when a note is pressed.
  Serial.print("Note on: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleNoteOff(byte channel, byte pitch, byte velocity) {
  // Log when a note is released.
  Serial.print("Note off: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleControlChange(byte channel, byte pitch, byte velocity) {
  // Log when a note is released.
  Serial.print("Control change: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

AudioOutput updateAudio(){
  return MonoOutput::from16Bit(1234);
}

void loop() {
  audioHook(); // required here
}
```

Here a diff of what I changed to make your life easier:
```
4a5,7
> #define MOZZI_CONTROL_RATE 128 // Hz, powers of 2 are most reliable
> #include <Mozzi.h>
> 
34a38,39
> 
>   startMozzi();
37c42
< void loop() {
---
> void updateControl() {
85a91,98
> 
> AudioOutput updateAudio(){
>   return MonoOutput::from16Bit(1234);
> }
> 
> void loop() {
>   audioHook(); // required here
> }
```

This is my platformio.ini
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
board_build.core = earlephilhower
board = rpipicow
monitor_speed = 9600
lib_deps =
	adafruit/Adafruit TinyUSB Library@^3.3.1
	sensorium/Mozzi@^2.0.0
build_flags =
	-DDEBUG_RP2040_PORT=Serial1
	-DUSE_TINYUSB
	-Iinclude/
	; -DCFG_TUSB_CONFIG_FILE=\"custom_tusb_config.h\"
	-DMOZZI_AUDIO_MODE=MOZZI_OUTPUT_I2S_DAC
	-DPIO_FRAMEWORK_ARDUINO_ENABLE_EXCEPTIONS
```

tfry-git, 2024-07-02:
Hi,

sorry for the delay. First idea: Your debug port appears to be Serial1. Isn't that pins 0, and 1, by default? That'd be the same as the default Mozzi output pins on RP2040. If so, try using either Serial2, instead, or change the Mozzi output pins (see https://sensorium.github.io/Mozzi/learn/configuration/ and https://sensorium.github.io/Mozzi/doc/html/hardware_rp2040.html).

leobel96, 2024-07-03:
Hello and thank you for your reply. I tried changing the debug port to Serial2 but nothing changed unfortunately 😞. Do you have any way of reproducing it on your side by any chance?

tfry-git, 2024-07-03:
> Hello and thank you for your reply. I tried changing the debug port to Serial2 but nothing changed unfortunately 😞. Do you have any way of reproducing it on your side by any chance?

I'll try. But how do I even go about this? MIDI is connected to the USB port, right, and then an FTDI attached for serial? To which pins?

leobel96, 2024-07-03:
No FTDI connected, the serialport enables the serial messages through the usb interface from my understanding https://arduino-pico.readthedocs.io/en/latest/serial.html

tfry-git, 2024-07-03:
Ok, but so where is MIDI attached in your setup? Can't have both on the same USB port, can you?

tomcombriat, 2024-07-03:
From the Doc I thought you would not need an ftdi, and do everything like it is done in teensy (with a select of how the usb should be seen by the computer) but I also do not have a clear view and no certitude, please enlighten :) 
I'm actually quite interested in having usb-midi on the Pico, and discovered this library with this issue. I'll try to give it a shot but I'm a bit away from breadboard at the moment...

leobel96, 2024-07-03:
Yeah, you can have both MIDI and Serial on USB. It works fine in the first example I posted without Mozzi

tomcombriat, 2024-07-03:
Random guess, before testing myself, have you tried putting Mozzi completely on the other core of the RP2040?

leobel96, 2024-07-03:
What do you mean with "other core", sorry?

leobel96, 2024-07-03:
Oh, I just noticed https://arduino-pico.readthedocs.io/en/latest/multicore.html, I'll give it a try :)

tomcombriat, 2024-07-03:
Big interest of the Pico :)

In brief, put all the calls for one library in loop and setup (like your first example), and for the other in setup1 and loop1. 

That's especially interesting for audio when you want the audio generation to run without any blockage caused by slow hardware or operations

leobel96, 2024-07-03:
So, I tried the following code:
```cpp
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

#define MOZZI_CONTROL_RATE 128 // Hz, powers of 2 are most reliable
#include <Mozzi.h>
#include <Oscil.h> // oscillator template
#include <tables/sin2048_int8.h> // sine table for oscillator
#include <mozzi_midi.h>
#include <ADSR.h>

#include "user.h"
#include "main.h"

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

// audio sinewave oscillator
Oscil <SIN2048_NUM_CELLS, MOZZI_AUDIO_RATE> aSin(SIN2048_DATA);

// envelope generator
ADSR <MOZZI_CONTROL_RATE, MOZZI_AUDIO_RATE> envelope;

void setup() {
  // Manual begin() is required on core without built-in support e.g. mbed rp2040
  if (!TinyUSBDevice.isInitialized()) {
    TinyUSBDevice.begin(0);
  }

  pinMode(14, OUTPUT);

  usb_midi.setStringDescriptor("TinyUSB MIDI");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);

  MIDI.setHandleNoteOn(handleNoteOn);
  MIDI.setHandleNoteOff(handleNoteOff);
  MIDI.setHandleControlChange(handleControlChange);

  Serial.begin(9600);
}

void setup1() {
  envelope.setADLevels(255,64);
  envelope.setTimes(50,200,10000,200); // 10000 is so the note will sustain 10 seconds unless a noteOff comes

  aSin.setFreq(440); // default frequency
  startMozzi();
  digitalWrite(14, HIGH);
}

void loop() {
  #ifdef TINYUSB_NEED_POLLING_TASK
  // Manual call tud_task since it isn't called by Core's background
  TinyUSBDevice.task();
  #endif

  // not enumerated()/mounted() yet: nothing to do
  if (!TinyUSBDevice.mounted()) {
    return;
  }
  // read any new MIDI messages
  MIDI.read();
}

void handleNoteOn(byte channel, byte pitch, byte velocity) {
  // Log when a note is pressed.
  Serial.print("Note on: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);

  aSin.setFreq(mtof(float(pitch)));
  envelope.noteOn();
  digitalWrite(14, LOW);
}

void handleNoteOff(byte channel, byte pitch, byte velocity) {
  // Log when a note is released.
  Serial.print("Note off: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);

  envelope.noteOff();
}

void handleControlChange(byte channel, byte pitch, byte velocity) {
  // Log when a note is released.
  Serial.print("Control change: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void updateControl(){
  // not enumerated()/mounted() yet: nothing to do
  if (!TinyUSBDevice.mounted()) {
    return;
  }
  envelope.update();
}

void loop1() {
  audioHook(); // required here
}

AudioOutput updateAudio(){
  return MonoOutput::from16Bit(envelope.next() * aSin.next());
}
```
And the MIDI port doesn't disappear anymore BUT the messages are still not being printed when a MIDI note is being received nor the LED is turned on (meaning that, probably, the execution never goes past the `startMozzi()` call).

If you want to quickly test MIDI, I suggest you to use the [mido](https://mido.readthedocs.io/en/stable/#) library for python and use the script I wrote to send simple commands:
```python
import mido

print(mido.get_output_names())

outport = mido.open_output("Pico W")

outport.send(mido.Message("control_change", control=3, value=10))
outport.send(mido.Message("control_change", control=4, value=100))
outport.send(mido.Message("control_change", control=5, value=127))
outport.send(mido.Message("control_change", control=6, value=40))

outport.send(mido.Message("note_on", note=60, velocity=40))
outport.send(mido.Message("note_on", note=61, velocity=40))
outport.send(mido.Message("note_on", note=62, velocity=40))
outport.send(mido.Message("note_on", note=63, velocity=40))
outport.send(mido.Message("note_on", note=64, velocity=40))
outport.send(mido.Message("note_on", note=65, velocity=40))
outport.send(mido.Message("note_on", note=66, velocity=40))
outport.send(mido.Message("note_on", note=67, velocity=40))

```

leobel96, 2024-07-03:
Actually, I just realised that I set the wrong serial port in my PC 🤦🏻‍♂️ and the prints work as well but I think that the library still has issues because everything I put in the second core after `startMozzi()` (either in the `setup1` itself or in `loop1`) never gets executed

tfry-git, 2024-07-04:
Ok, I finally tried, and after some initial trouble, it all seems to work for me without trouble (single core variant).

I.e. literally the sketch from your initial post (the one including Mozzi, of course) works for me, with the following differences:
- I had to remove the two lines ```#include "user.h"``` and ```#include "main.h"```. (Not sure, if that's platformio thing, I'm testing with the plain Arduino IDE, here).
- The port gets listed as "Pico MIDI 1", not just "Pico", for me

It may be noteworthy, that after some initial experiments,  I updated my rp2040 core to version 3.9.3 (don't remember which version I had, before). Before that, independent of Mozzi, no MIDI port got registered without an explicit call to ```usb_midi.setCableName()```. I don't know what's going on here, but I do suggest one of the first things to try is to update the rp2040 core.

leobel96, 2024-07-06:
`#include "user.h"` and `#include "main.h"` were just containing the definitions of the functions I declared in the code.
I tried using the Arduino IDE as you suggested and everything is working as expected! I don't know why I'm seeing those issues with Platformio and I tried updating the RP2040 core there as well but I cannot make it work. I think I'll stick with the Arduino IDE for now. Thanks a lot for your help!
About the dual core approach, moving the `TinyUSBDevice.mounted()` to the core0 (i.e. `loop()`) should work (I should have put it there from the beginning tbf).

leobel96, 2024-09-08:
Hello, unfortunately I need to reopen this issue: I went back to this project after a while and I managed to buy an I2S DAC. Again, using the latest versions of all the libraries and the Arduino IDE, I cannot make the following script work:

```cpp
#define MOZZI_AUDIO_MODE MOZZI_OUTPUT_I2S_DAC

// all of these are the defaults on RP2040 outputting on I2S, anyway, thus commented
#define MOZZI_I2S_PIN_BCK 16
#define MOZZI_I2S_PIN_WS (MOZZI_I2S_PIN_BCK+1) // CANNOT BE CHANGED, HAS TO BE NEXT TO pBCLK, i.e. default is 21
#define MOZZI_I2S_PIN_DATA 18

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// Create the I2S port using a PIO state machine
#include <Mozzi.h>
#include <Oscil.h> // oscillator template
#include <tables/sin2048_int8.h> // sine table for oscillator

// use: Oscil <table_size, update_rate> oscilName (wavetable), look in .h file of table #included above
Oscil <SIN2048_NUM_CELLS, MOZZI_AUDIO_RATE> aSin(SIN2048_DATA);

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void handleNoteOn(byte channel, byte pitch, byte velocity) {
  Serial.printf("Note on: channel = %d pitch = %d velocity = %d", channel, pitch, velocity);
}

void setup() {
  if (!TinyUSBDevice.isInitialized()) {
    TinyUSBDevice.begin(0);
  }

  Serial.begin(115200);

  usb_midi.setStringDescriptor("TinyUSB MIDI");

  MIDI.begin(MIDI_CHANNEL_OMNI);

  if (TinyUSBDevice.mounted()) {
    TinyUSBDevice.detach();
    delay(10);
    TinyUSBDevice.attach();
  }
  // MIDI.turnThruOff();
  MIDI.setHandleNoteOn(handleNoteOn);
  Serial.print("Done");
}

void loop(){
  #ifdef TINYUSB_NEED_POLLING_TASK
  // Manual call tud_task since it isn't called by Core's background
  TinyUSBDevice.task();
  #endif

  // not enumerated()/mounted() yet: nothing to do
  if (!TinyUSBDevice.mounted()) {
    return;
  }
  MIDI.read();
}

void setup1(){
  startMozzi(); // :)
  aSin.setFreq(440); // set the frequency
}

void updateControl(){
  // put changing controls in here
}

AudioOutput updateAudio(){
  return MonoOutput::from8Bit(aSin.next()); // return an int signal centred around 0
}

void loop1(){
  audioHook(); // required here
}

```
It looks, again, like some kind of interaction between tinyUSB/MIDI and Mozzi because I'm able to hear the 440Hz sound when I remove all the tinyUSB/MIDI stuff:
```cpp
#define MOZZI_AUDIO_MODE MOZZI_OUTPUT_I2S_DAC

// all of these are the defaults on RP2040 outputting on I2S, anyway, thus commented
#define MOZZI_I2S_PIN_BCK 16
#define MOZZI_I2S_PIN_WS (MOZZI_I2S_PIN_BCK+1) // CANNOT BE CHANGED, HAS TO BE NEXT TO pBCLK, i.e. default is 21
#define MOZZI_I2S_PIN_DATA 18

#include <Arduino.h>
// #include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// Create the I2S port using a PIO state machine
#include <Mozzi.h>
#include <Oscil.h> // oscillator template
#include <tables/sin2048_int8.h> // sine table for oscillator

// use: Oscil <table_size, update_rate> oscilName (wavetable), look in .h file of table #included above
Oscil <SIN2048_NUM_CELLS, MOZZI_AUDIO_RATE> aSin(SIN2048_DATA);

// USB MIDI object
// Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
// MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void handleNoteOn(byte channel, byte pitch, byte velocity) {
  Serial.printf("Note on: channel = %d pitch = %d velocity = %d", channel, pitch, velocity);
}

void setup() {
  Serial.begin(115200);
  // MIDI.begin(MIDI_CHANNEL_OMNI);
  // MIDI.turnThruOff();
  // MIDI.setHandleNoteOn(handleNoteOn);
  Serial.print("Done");
}

void loop(){
  // MIDI.read();
}

void setup1(){
  startMozzi(); // :)
  aSin.setFreq(440); // set the frequency
}

void updateControl(){
  // put changing controls in here
}

AudioOutput updateAudio(){
  return MonoOutput::from8Bit(aSin.next()); // return an int signal centred around 0
}

void loop1(){
  audioHook(); // required here
}

```

tomcombriat, 2024-09-09:
Sorry for closing prematurely. Will also try to give it a shot soon!

tomcombriat, 2024-09-17:
Alright, finally found a bit of time to test here. Sorry for the delay.

At first glance, all works okay on my setup. I changed slightly the I2S settings to adapt to a synth I had at hand. Here is my sketch:

      #define MOZZI_AUDIO_MODE MOZZI_OUTPUT_I2S_DAC
      #define MOZZI_I2S_FORMAT MOZZI_I2S_FORMAT_LSBJ
      
      // all of these are the defaults on RP2040 outputting on I2S, anyway, thus commented
      #define MOZZI_I2S_PIN_BCK 20
      #define MOZZI_I2S_PIN_WS (MOZZI_I2S_PIN_BCK+1) // CANNOT BE CHANGED, HAS TO BE NEXT TO pBCLK, i.e. default is 21
      #define MOZZI_I2S_PIN_DATA 22
      
      #include <Arduino.h>
      #include <Adafruit_TinyUSB.h>
      #include <MIDI.h>
      
      // Create the I2S port using a PIO state machine
      #include <Mozzi.h>
      #include <Oscil.h> // oscillator template
      #include <tables/sin2048_int8.h> // sine table for oscillator
      
      // use: Oscil <table_size, update_rate> oscilName (wavetable), look in .h file of table #included above
      Oscil <SIN2048_NUM_CELLS, MOZZI_AUDIO_RATE> aSin(SIN2048_DATA);
      
      // USB MIDI object
      Adafruit_USBD_MIDI usb_midi;
      
      // Create a new instance of the Arduino MIDI Library,
      // and attach usb_midi as the transport.
      MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);
      
      void handleNoteOn(byte channel, byte pitch, byte velocity) {
        Serial.printf("Note on: channel = %d pitch = %d velocity = %d", channel, pitch, velocity);
        digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
      }
      
      void setup() {
        pinMode(LED_BUILTIN,OUTPUT);
        if (!TinyUSBDevice.isInitialized()) {
          TinyUSBDevice.begin(0);
        }
      
        Serial.begin(115200);
      
        usb_midi.setStringDescriptor("TinyUSB MIDI");
      
        MIDI.begin(MIDI_CHANNEL_OMNI);
      
        if (TinyUSBDevice.mounted()) {
          TinyUSBDevice.detach();
          delay(10);
          TinyUSBDevice.attach();
        }
        // MIDI.turnThruOff();
        MIDI.setHandleNoteOn(handleNoteOn);
        Serial.print("Done");
      }
      
      void loop(){
        #ifdef TINYUSB_NEED_POLLING_TASK
        // Manual call tud_task since it isn't called by Core's background
        TinyUSBDevice.task();
        #endif
      
        // not enumerated()/mounted() yet: nothing to do
        if (!TinyUSBDevice.mounted()) {
          return;
        }
        MIDI.read();
      }
      
      
      void setup1(){
        startMozzi(); // :)
        aSin.setFreq(440); // set the frequency
      }
      
      void updateControl(){
        // put changing controls in here
      }
      
      AudioOutput updateAudio(){
        return MonoOutput::from8Bit(aSin.next()); // return an int signal centred around 0
      }
      
      void loop1(){
        audioHook(); // required here
      }

As you can see, I just changed the I2S config for my setup, and using a PT8211 dac which requires Japanese I2S format, and a small blinker on the built-in led to acknowledge reception of a noteOn signal on the USB midi port.

---

## Test condition:
  - IDE: Arduino v2.3.2
  - Core: https://github.com/earlephilhower/arduino-pico, v4.0.2
  - Mozzi: github version
  - OS: Debian  11 (GNU/Linux)
  - TinyUSB: v3.3.3
  - MIDI: connections with ALSA, MIDI generated with VMPK. The created MIDI port actually shows as `0:Pico MIDI 1` -> same than @tfry-git (why?)
  - AUDIO: standard oscilloscope wired to the output of the synth (which is https://github.com/tomcombriat/G_TES_touch-synth to be exact).
  - Testing with your configuration does not lead to any sound (because the DAC is hardwired differently), but MIDI input is acknowledge.
  - USB stack set to: Adafruit TinyUSB (not the native option).

Do not know what to test more… What is you exact configuration? Are you on the Arduino IDE of PlatformIO? Are you on GNU/Linux or mac or Windows?

leobel96, 2024-09-19:
Uhm, I tried using the exact same versions as in your "Test condition" and also moving the DAC to use the exact same pins as you without luck (I even tried using exactly the same snippet you posted). I'm on Mac. Could you please check if you have exactly this configuration in the Arduino IDE?
[image]
The behaviour I'm seeing is that the MIDI connection is working as expected but no sound is played. Also, repeating it here in case it got lost in the messages, I'm using a Raspberry Pico W (the version with the WiFi).

tomcombriat, 2024-09-19:
Here is my config, plus a show that a sound is played (see Audacity) and the MIDI port appears in ALSA.
[image]

So now, why does not it work on your side? A few thoughts:
 - you are using the Pico W (I indeed overlooked that)
 - have you tried without a DAC? Using standard PWM output?
 - Are you sending midi signals straight on boot up of the RP?

leobel96, 2024-10-16:
Hello, sorry for the delay but I just had a chance to put my hands on the project again. I'm not sending MIDI signals straight on boot. 
I tried using the PWM out as you suggested and it looks like it's working fine:
[image]
and the MIDI still works as well! 
It looks like, indeed, it's something DAC-related

Just a curiosity: how do you connect the PWM to Audacity? Do you use some custom circuit/board or you connect the audio jack input straight to the pins of the RP2040? Do you need some voltage regulation?

tomcombriat, 2024-10-16:
That's a good start!

> It looks like, indeed, it's something DAC-related

Indeed, but if you are still using the Pico Wifi, it seems that it is a compatibility problem between this chip and I2S as I managed to get an I2S dac working with Mozzi and TinyUSB (what DAC are you using by the way?)

> Just a curiosity: how do you connect the PWM to Audacity? Do you use some custom circuit/board or you connect the audio jack input straight to the pins of the RP2040? Do you need some voltage regulation?

I plug it straight into an external Audio interface (with gain control). The great advantage is that the audio interface filters the PWM frequency out and hence display only the audio part. Otherwise, for instance with a DAC, I prefer using a scope but that means adding a low-pass filter with PWM output otherwise you only see the PWM carrier!

leobel96, 2024-10-16:
> it seems that it is a compatibility problem between this chip and I2S

I could try generating a simple tone on the I2S interface without Mozzi and while using the MIDI interface too

> what DAC are you using by the way?

I'm using a [PCM5102](https://www.aliexpress.com/item/1005006104038963.html)

> Would you have any way to test with a non-wifi rp2040?

No unfortunately :(

> I plug it straight into an external Audio interface (with gain control)

That makes sense! What external audio interface are you using?

leobel96, 2024-10-16:
Okay, with the following snippet I can both hear a sound on the DAC **and** receive MIDI messages:

```cpp
/*
  This example generates a square wave based tone at a specified frequency
  and sample rate. Then outputs the data using the I2S interface to a
  MAX08357 I2S Amp Breakout board.

  created 17 November 2016
  by Sandeep Mistry
  modified for RP2040 by Earle F. Philhower, III <earlephilhower@yahoo.com>

    bool setBCLK(pin_size_t pin);
    - This assigns two adjacent pins - the pin after this one (one greater)
      is the WS (word select) signal, which toggles before the sample for
      each channel is sent

    bool setDATA(pin_size_t pin);
    - Sets the DOUT pin, can be any valid GPIO pin
*/

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

#include <I2S.h>

// Create the I2S port using a PIO state machine
I2S i2s(OUTPUT);

// GPIO pin numbers
#define pBCLK 20
#define pWS (pBCLK+1)
#define pDOUT 22

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

const int frequency = 440; // frequency of square wave in Hz
const int amplitude = 500; // amplitude of square wave
const int sampleRate = 16000; // minimum for UDA1334A

const int halfWavelength = (sampleRate / frequency); // half wavelength of square wave

int16_t sample = amplitude; // current sample value
int count = 0;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, 1);

  Serial.begin(115200);
  Serial.println("I2S simple tone");

  i2s.setBCLK(pBCLK);
  i2s.setDATA(pDOUT);
  i2s.setBitsPerSample(16);

  // start I2S at the sample rate with 16-bits per sample
  if (!i2s.begin(sampleRate)) {
    Serial.println("Failed to initialize I2S!");
    while (1); // do nothing
  }

  if (!TinyUSBDevice.isInitialized()) {
    TinyUSBDevice.begin(0);
  }

  usb_midi.setStringDescriptor("TinyUSB MIDI");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MIDI.begin(MIDI_CHANNEL_OMNI);

  // If already enumerated, additional class driverr begin() e.g msc, hid, midi won't take effect until re-enumeration
  if (TinyUSBDevice.mounted()) {
    TinyUSBDevice.detach();
    delay(10);
    TinyUSBDevice.attach();
  }

  // Attach the handleNoteOn function to the MIDI Library. It will
  // be called whenever the Bluefruit receives MIDI Note On messages.
  MIDI.setHandleNoteOn(handleNoteOn);

  // Do the same for MIDI Note Off messages.
  MIDI.setHandleNoteOff(handleNoteOff);

}

void loop() {
  if (count % halfWavelength == 0) {
    // invert the sample every half wavelength count multiple to generate square wave
    sample = -1 * sample;
  }

  // write the same sample twice, once for left and once for the right channel
  i2s.write(sample);
  i2s.write(sample);

  // increment the counter for the next sample
  count++;

  #ifdef TINYUSB_NEED_POLLING_TASK
  // Manual call tud_task since it isn't called by Core's background
  TinyUSBDevice.task();
  #endif

  // not enumerated()/mounted() yet: nothing to do
  if (!TinyUSBDevice.mounted()) {
    return;
  }

  MIDI.read();
}

void handleNoteOn(byte channel, byte pitch, byte velocity) {
  // Log when a note is pressed.
  Serial.print("Note on: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

void handleNoteOff(byte channel, byte pitch, byte velocity) {
  // Log when a note is released.
  Serial.print("Note off: channel = ");
  Serial.print(channel);

  Serial.print(" pitch = ");
  Serial.print(pitch);

  Serial.print(" velocity = ");
  Serial.println(velocity);
}

```

so it looks like it's something specific to Mozzi to me

tomcombriat, 2024-10-20:
Hi again!

Sorry for the delay. 

> so it looks like it's something specific to Mozzi to me

Starts indeed to be that way, even though Mozzi does little more than initializing the I2S just the way your last sketch is doing. If I am correct, your [last attempt](https://github.com/sensorium/Mozzi/issues/263#issuecomment-2336667040) with Mozzi/I2S was using pin 16/17/18 for the I2S, whereas you got it to work without Mozzi with pins 20,21 and 22. Have you tried with that other set of pins and Mozzi? 
If so, only the board is different between you and me (I am not using a "W" pico version, will try to get my hands on one at some point). The fact that PWM works is reassuring and seems to point toward a I2S/Mozzi/TinyUSB conflict somehow.

Just to make sure, could you try deactivating Mozzi's async analogRead? Ie, adding `#define MOZZI_ANALOG_READ MOZZI_ANALOG_READ_NONE`  in your config at the beginning of the sketch? That will disable the only other Mozzi function that are doing "something fancy" (using DMA and so on).

---

> That makes sense! What external audio interface are you using?

A fairly old Alesis iO4. Not the quiestest but I really make use of the standard MIDI connections it does provide ;)

leobel96, 2024-10-21:
> If I am correct, your last attempt with Mozzi/I2S was using pin 16/17/18 for the I2S, whereas you got it to work without Mozzi with pins 20,21 and 22. Have you tried with that other set of pins and Mozzi?

I have been testing using the same pins you use (20, 21, 22) since that time and I can confirm that they behave exactly like (16, 17, 18).

> could you try deactivating Mozzi's async analogRead? Ie, adding #define MOZZI_ANALOG_READ MOZZI_ANALOG_READ_NONE in your config at the beginning of the sketch?

Sure, I'll try in the next couple of days. 

I also got my hands on a CMSYS-DAP debugger, so I can try with some breakpoints to see where it actually gets stuck

leobel96, 2024-10-22:
Breaking news: I just tried adding `#define MOZZI_ANALOG_READ MOZZI_ANALOG_READ_NONE ` and everything works!

tomcombriat, 2024-10-23:
Ah! Nice! So we know that the conflict is probably somewhere between Mozzi's async read and TinyUSB, on the "W" version of the RP2040. Considering the fact that there the number of ADC in the RP is quite low, and that they can always be run on the other core to avoid delay in Mozzi I think that is not such a big deal. Will try to find a place to state it though as it can be annoying for some that are not aware. If I get my hands (or someone else) on one of these, I will try to dive into it.

Can I close this?

leobel96, 2024-10-23:
> Considering the fact that there the number of ADC in the RP is quite low, and that they can always be run on the other core to avoid delay in Mozzi I think that is not such a big deal

Do you mean suggesting to use the normal `AnalogRead` instead of `mozziAnalogRead`? If yes, how slow would it be? I'm asking because in my sketch I'm also trying to use the value of an analog input in the `updateControl()`

tomcombriat, 2024-10-23:
> Do you mean suggesting to use the normal AnalogRead instead of mozziAnalogRead?

Well, Mozzi defaults `mozziAnalogRead` to a standard `analogRead` in that case, so it does not matter (you can keep the Mozzi version in case we manage to fix this.).

>  If yes, how slow would it be?

Hard to say, I never timed it. But except with a super high load, or fairly high `CONTROL_RATE` that should probably not be a problem. I suggest just trying. If you run into underrun problems, it is always possible to run the ADC on the other core (if that core is not busy doing something long and blocking) for instance (not compiled, just for the example):

``` c++
#include <Mozzi.h>
#include <Oscil.h> // oscillator template
#include <tables/sin2048_int8.h> // sine table for oscillator

Oscil <SIN2048_NUM_CELLS, MOZZI_AUDIO_RATE> aSin(SIN2048_DATA);

/*volatile */ uint8_t gain=0;;  // I am never sure if this actually needs to be volatile, I never had problems without
unsigned long time_between_reads=3, last_read_time=0; // in milliseconds
int pin_adc = 26;

void setup(){
  startMozzi(); // start with default control rate of 64
  aSin.setFreq(3320); // set the frequency
}

void setup1() // other core
{
pinMode(pin_adc, INPUT);
}

void loop1()
{
if (millis() > last_read_time+time_between_reads)
{
gain = mozziAnalogRead<8>(pin_adc);
last_read_time=millis();
}
}

void updateControl(){ 
}

AudioOutput updateAudio(){
  return MonoOutput::from16Bit(aSin.next() * gain); // 8 bits waveform * 8 bits gain makes 16 bits
}

void loop(){
  audioHook(); // required here
}
```

closed 2024-11-19 by leobel96