sensorium/Mozzi #264: Mozzi envelopes produce long note

justinjools22, 2024-06-30:
Hi, I have been playing around with the ADSR_Audio_Rate_Envelope example. I thought by setting the attack, decay, sustain, release manually with values e.g. 100, 1000, 1000, 500 I would be able to produce a long string like sound that lasts for a few seconds but it produces a jittery sound. So it seems that is some other parameter at play. I would just like to emulate a long note to understand how it works. Is there something like Envelope.NoteHold or length/time?

tomcombriat, 2024-06-30:
Hi! 
The envelopes come with a third template parameter which is useful for producing long notes, have you tried [this example? []()](https://github.com/sensorium/Mozzi/blob/master/examples/07.Envelopes/ADSR_Audio_Rate_Envelope_Long/ADSR_Audio_Rate_Envelope_Long.ino)

justinjools22, 2024-06-30:
I was just playing around with adding in static ADSR values but just produces a long consistent drone. My code:

#include "MozziConfigValues.h"  // for named option values
#define MOZZI_AUDIO_MODE MOZZI_OUTPUT_EXTERNAL_TIMED
#define MOZZI_CONTROL_RATE 256 // Hz, powers of 2 are most reliable

#include <Mozzi.h>
#include <Oscil.h>
#include <EventDelay.h>
#include <ADSR.h>
#include <tables/sin8192_int8.h>
// #include <mozzi_rand.h>
#include <mozzi_midi.h>
// #include <mozzi_fixmath.h>
#include <DAC_MCP49xx.h>  // https://github.com/tomcombriat/DAC_MCP49XX 

Oscil <8192, MOZZI_AUDIO_RATE> aOscil(SIN8192_DATA);;

// External audio output parameters and DAC declaration
#define SS_PIN PB1  // if you are on AVR and using PortWrite you need still need to put the pin you are actually using: 7 on Uno, 38 on Mega
#define BITS_PER_CHANNEL 12  // each channel of the DAC is outputting 12 bits
// #define LDAC_PIN PB10 // To simultaneously update 2x DAC's
DAC_MCP49xx dac(DAC_MCP49xx::MCP4922, SS_PIN);

// for triggering the envelope
EventDelay noteDelay;

ADSR <MOZZI_AUDIO_RATE, MOZZI_AUDIO_RATE, unsigned long> envelope;

boolean note_is_on = true;

void audioOutput(const AudioOutput f)
{
  uint16_t out = (f.l() >> 4) + 2048;
  dac.output(out);
}

void setup(){
  dac.init();
  //Serial.begin(9600); // for Teensy 3.1, beware printout can cause glitches
  Serial.begin(115200);
  randSeed(); // fresh random
  noteDelay.set(2000); // 2 second countdown
  startMozzi();
}

unsigned int duration, attack, decay, sustain, release_ms;

void updateControl(){
  if(noteDelay.ready()){
      // choose envelope levels
        byte attack_level = 100;
        byte decay_level = 100;
      envelope.setADLevels(attack_level,decay_level);

       attack = 50; 
       decay = 200; 
       sustain = 200;
       release_ms = 1000;
     }
     envelope.setTimes(attack,decay,sustain,release_ms);
     envelope.noteOn();

//     byte midi_note = rand(107)+20;
     byte midi_note = 50;     
     aOscil.setFreq((int)mtof(midi_note));
     noteDelay.start(attack+decay+sustain+release_ms);
   }
//}

AudioOutput updateAudio(){
  envelope.update();
  return MonoOutput::fromNBit(12, (int32_t) (envelope.next() * aOscil.next()) );
}

void loop(){
  audioHook(); // required here
}

tomcombriat, 2024-06-30:
You can also try to play with the parameter of noteOn(https://sensorium.github.io/Mozzi/doc/html/class_a_d_s_r.html#aef8d9c2799485fcd5a854228503f9f2f), resetting the phase of the envelope might be what you are looking for?

tomcombriat, 2024-07-03:
Hi! 
Is this solved?

tomcombriat, 2024-07-29:
Closing this. If this is not solved do not hesitate to re-open.

closed 2024-07-29 by tomcombriat