sensorium/Mozzi #268: 04.Audio_Input error 'AUDIO_INPUT_PIN' was not declared in this scope

justinjools22, 2024-07-26:
Hi, I tried to compile this code as is and get error. I am using Windows 11.

In file included from c:\Users\m\Documents\Arduino\libraries\Mozzi/MozziGuts.h:205,
                 from c:\Users\m\Documents\Arduino\libraries\Mozzi/Mozzi.h:33,
                 from C:\Users\m\AppData\Local\Temp\.arduinoIDE-unsaved2024626-3808-2cxc72.nn99x\Audio_Input\Audio_Input.ino:29:
c:\Users\m\Documents\Arduino\libraries\Mozzi/internal/MozziGuts.hpp: In function 'void MozziPrivate::advanceADCStep()':
c:\Users\m\Documents\Arduino\libraries\Mozzi/internal/MozziGuts.hpp:175:43: error: 'AUDIO_INPUT_PIN' was not declared in this scope; did you mean 'MOZZI_AUDIO_INPUT_PIN'?
  175 |     adcStartConversion(adcPinToChannelNum(AUDIO_INPUT_PIN));  // -> result is ignored, but first thing in the next cycle, a second reading is taken.
      |                                           ^~~~~~~~~~~~~~~
      |                                           MOZZI_AUDIO_INPUT_PIN

exit status 1

Compilation error: exit status 1

https://github.com/sensorium/Mozzi/blob/master/examples/04.Audio_Input/Audio_Input/Audio_Input.ino

justinjools22, 2024-07-26:
I looked in Mozziguts.hh from the error at 175. The sketches need to change:

#define MOZZI_AUDIO_INPUT_PIN 0 to
#define AUDIO_INPUT_PIN 0

    adcStartConversion(adcPinToChannelNum(AUDIO_INPUT_PIN));  // -> result is ignored, but first thing in the next cycle, a second reading is taken.
    break;

tomcombriat, 2024-07-26:
Or the other way around? 

This comes from the changes to Mozzi2, trying to make internal variables less likely to conflict. Please submit a PR if you solved it :)

tfry-git, 2024-07-27:
@tomcombriat You fixed that one a month ago, yourself: https://github.com/sensorium/Mozzi/commit/fdd36681c08e2f5306fa8a2cb1080198904df58a

Perhaps we should have a quick bugfix release?

tomcombriat, 2024-07-27:
Ah... Aging.... And I thought it rang a bell when I saw it.

Yes, maybe we should..

justinjools22, 2024-07-27:
> Or the other way around?
> 
> This comes from the changes to Mozzi2, trying to make internal variables less likely to conflict. Please don't hesitate submit a PR if you solved it :)

Yes, I guess your naming intention is MOZZI_AUDIO_INPUT_PIN. Because it is incorrectly names AUDIO_INPUT_PIN works.

tfry-git, 2024-07-28:
> Yes, maybe we should..

You are more experienced in this step. Could you take a look at my draft: https://github.com/sensorium/Mozzi/releases/tag/untagged-64bc2f81715cb7b13227 . Anything else missing?

tomcombriat, 2024-07-29:
> Could you take a look at my draft: https://github.com/sensorium/Mozzi/releases/tag/untagged-64bc2f81715cb7b13227 . Anything else missing?

Looks good to me! I will publish it.

tomcombriat, 2024-07-29:
Closing. Fixed by #260

closed 2024-07-29 by tomcombriat