sensorium/Mozzi #269: RP2040 PlatformIO lacks PWMAudio dependency

kinoshita-lab, 2024-08-07:
## Environment:
- Windows 11 23H2
- PlatformIO Core, version 6.1.15
- Mozzi 2.0.1
- Board: RP2040

## Steps to reproduce

1.  Create new project with `Board:RP2040 (Generic)`  and `Framework: Arduino`
2. Tested with minimum code below for main.cpp:
```.cpp
#include <Arduino.h>
#include <Mozzi.h>

void setup() {}

void loop() {}
```

3. execute "PlatformIO: Build"

## Current Behavior
Resulting to following error:
```.sh
In file included from .pio\libdeps\generic\Mozzi/internal/MozziGuts.hpp:59,
                 from .pio\libdeps\generic\Mozzi/MozziGuts.h:205,
                 from .pio\libdeps\generic\Mozzi/Mozzi.h:33,
                 from src\main.cpp:2:
.pio\libdeps\generic\Mozzi/internal/MozziGuts_impl_RP2040.hpp:162:9: fatal error: PWMAudio.h: No such file or directory
```

## Workaround
Add `PWMAudio` to the project's `platformio.ini`, for example it looks like below:
```.ini
lib_deps = 
    sensorium/Mozzi@^2.0.1
    PWMAudio
```

## Comments
Mozzi for PlatformIO already has dependency with `I2S` and `AudioBufferManager`, but it lacks `PWMAudio`.
This issue would be solved by modifing `library.json`, but I didn't find it so far.

tomcombriat, 2024-08-07:
Hi,
Thanks for reporting this! As far as I know, no-one in the development team is using PlatformIO with Mozzi, even though some users have reported using it.

If I am correct, there is no `library.json` for Mozzi. Actually, it was proposed some time ago (#15) but was not pursued…

One thing I wonder (but maybe you can enlighten us on that) is, as Mozzi supports a quite large set of different architectures, all with different includes, is it possible to make this `library.json` specific enough to have different dependencies per arch? Interestingly, PlatformIO knew that `I2S` and `AudioBufferManager` were needed…

tfry-git, 2024-08-07:
I'd have to check the timing, but AFAIR, PWMAudio was added to the RP2040 core about a year or two ago (https://github.com/earlephilhower/arduino-pico/tree/master/libraries/PWMAudio). So this looks like a problem of PlatformIO not having the latest version.

I have no real idea just how cores and libraries are registered with PlatformIO, but the problem will be found, there.

kinoshita-lab, 2024-08-08:
Hello, thank you for responding.

> I have no real idea just how cores and libraries are registered with PlatformIO, but the problem will be found, there.
I think so too, this issue can be solved by correcting PlatformIO library registration info about Mozzi.

According to the registry website, it shows authors of the library. Is it NOT the actual submitter who have registered Mozzi library over there? 

https://registry.platformio.org/libraries/sensorium/Mozzi/insights/authors

tfry-git, 2024-08-08:
I don't think we have registered that, ourselves, but the info looks correct.

However, I could not find the core we are assuming (https://github.com/earlephilhower/arduino-pico) in that registry. Rather it seems to contain a totally different, mbed-based core (https://github.com/platformio/platform-raspberrypi).

No idea how you would go about installing the former with PlatformIO (or about registering it in the database), but I think that's what you'll need to do.

tomcombriat, 2024-08-08:
> I don't think we have registered that, ourselves, but the info looks correct.

I don't think either. Also note the FixMath is also correctly registered even though I am sure of not doing it.

kinoshita-lab, 2024-08-18:
I added Mozzi manually by copying Mozzi files into src directory.
The problem still happens.
It seems the root cause of the problem is PlatformIO's automatic dependency detection.
I think here is not suitable place to report issue, while the workaround is still effective..

So, I close this issue now. 
Thank you for all of your cooperation!

closed 2024-08-18 by kinoshita-lab