sensorium/Mozzi #273: Running the most basic program on Pico rp2040 resulted in the following error:

lechenghhh, 2024-08-23:
In file included from d:\�ҵ��ĵ�\Arduino\libraries\Mozzi/MozziGuts.h:205:0,
                 from d:\�ҵ��ĵ�\Arduino\libraries\Mozzi/Mozzi.h:33,
                 from D:\我的文档\Arduino\Eurorack_PGM\Mozzi2040\Mozzi2040.ino:2:
d:\�ҵ��ĵ�\Arduino\libraries\Mozzi/internal/MozziGuts.hpp: In function 'void MozziPrivate::bufferAudioOutput(AudioOutput)':
d:\�ҵ��ĵ�\Arduino\libraries\Mozzi/internal/MozziGuts.hpp:85:3: error: 'audioOutput' was not declared in this scope
   audioOutput(f);
   ^~~~~~~~~~~
d:\�ҵ��ĵ�\Arduino\libraries\Mozzi/internal/MozziGuts.hpp:85:3: note: suggested alternative: 'AudioOutput'
   audioOutput(f);
   ^~~~~~~~~~~
   AudioOutput
d:\�ҵ��ĵ�\Arduino\libraries\Mozzi/internal/MozziGuts.hpp: In function 'void MozziPrivate::audioHook()':
d:\�ҵ��ĵ�\Arduino\libraries\Mozzi/internal/MozziGuts.hpp:232:7: error: 'canBufferAudioOutput' was not declared in this scope
   if (canBufferAudioOutput()) {
       ^~~~~~~~~~~~~~~~~~~~
d:\�ҵ��ĵ�\Arduino\libraries\Mozzi/internal/MozziGuts.hpp:232:7: note: suggested alternative: 'bufferAudioOutput'
   if (canBufferAudioOutput()) {
       ^~~~~~~~~~~~~~~~~~~~
       bufferAudioOutput

exit status 1

Compilation error: exit status 1

Excuse me, is it caused by some incorrect configurations of mine? The same program ran normally on Arduino Nano before

tomcombriat, 2024-08-23:
Hi!

There is too little information to spot the problem for now, but the answers to these questions should help :) :
 - What version of Mozzi are you using (Github, Arduino Library Manager), including the version number.
 - Have you tried running some of the examples sketch of Mozzi, with no change?
 - Could you provide a minimal non-working code?

lechenghhh, 2024-08-24:
Mozzi Version:2.0.1
Pgm: Mozzi/01.Basics/Sinewave.ino
Controller: Pico RP2040

tomcombriat, 2024-08-24:
Works here with no problem. Are you trying to compile the example with no modification? Have you tried any other examples? Are you using the Arduino IDE with Earle F. Philhower's core for RP2040?

lechenghhh, 2024-08-26:
arduino ide:2.0.5
boards:Arduino Mbed OS RP2040 Boards by Arduino  version 4.1.5

I also received the same prompt when running other mozzi example tutorials
I entered the 85th line of MozziGuts.hpp based on my error debugging information for audioOutput (f); This code is annotated
Then go to line 232 of MozziGuts.hpp and directly change the if (canBufferAudioOutput()) to if (true) code to enable normal compilation
I suspect that my microcontroller controller may have made an error when judging the configuration file of Mozzi

lechenghhh, 2024-08-26:
Before manually modifying the dependency libraries mentioned above, I can ensure that they are all the latest dependency libraries of Mozi
I want to use PWM to output Mozzi audio. Is i2s DAC enabled by default in some cases?
But adding macro definitions for PWM output has no effect, and there is too little information about developers who use Pico RP2040 and run Mozzi to find relevant materials

tomcombriat, 2024-08-27:
Hi,

Hi got some more comments here by email, but I cannot find them on the thread… Deleted?

In any case, did you manage? From what I saw, I think you were not using the correct core:
> boards:Arduino Mbed OS RP2040 Boards by Arduino version 4.1.5

whereas Mozzi has been ported to Earle F. Philhower's core

lechenghhh, 2024-08-27:
I am not sure about the relationship between this kernel and the default options I found in the Arduino IDE development board manager. However, how to use Earle F. Philhower's core on Pico rp2040? Maybe I should check out this information

lechenghhh, 2024-08-28:
Hello! After I switched to the development board core you mentioned, my rp2040 can now produce sound. Thank you very much for your question！

lechenghhh, 2024-08-28:
Hello! After I switched to the development board core you mentioned, my rp2040 can now produce sound. Thank you very much for your question

closed 2024-08-28 by lechenghhh