sensorium/Mozzi #276: all the example sketches dont compile on rp2040 nano

xxzykw, 2024-09-20:
In file included from /root/Arduino/libraries/Mozzi/MozziGuts.h:31:0,
                 from /root/Arduino/libraries/Mozzi/Mozzi.h:33,
                 from /root/Arduino/libraries/Mozzi/examples/01.Basics/Sinewave/Sinewave.ino:22:
/root/Arduino/libraries/Mozzi/internal/config_checks_generic.h:147:6: warning: #warning Asynchronous analog reads not implemented on this platform [-Wcpp]
 #    warning Asynchronous analog reads not implemented on this platform
      ^~~~~~~
In file included from /root/Arduino/libraries/Mozzi/internal/MozziGuts.hpp:59:0,
                 from /root/Arduino/libraries/Mozzi/MozziGuts.h:205,
                 from /root/Arduino/libraries/Mozzi/Mozzi.h:33,
                 from /root/Arduino/libraries/Mozzi/examples/01.Basics/Sinewave/Sinewave.ino:22:
/root/Arduino/libraries/Mozzi/internal/MozziGuts_impl_RP2040.hpp:298:2: warning: #warning Automatic random seeding is not implemented on this platform [-Wcpp]
 #warning Automatic random seeding is not implemented on this platform
  ^~~~~~~
In file included from /root/Arduino/libraries/Mozzi/MozziGuts.h:205:0,
                 from /root/Arduino/libraries/Mozzi/Mozzi.h:33,
                 from /root/Arduino/libraries/Mozzi/examples/01.Basics/Sinewave/Sinewave.ino:22:
/root/Arduino/libraries/Mozzi/internal/MozziGuts.hpp: In function 'void MozziPrivate::bufferAudioOutput(AudioOutput)':
/root/Arduino/libraries/Mozzi/internal/MozziGuts.hpp:85:3: error: 'audioOutput' was not declared in this scope
   audioOutput(f);
   ^~~~~~~~~~~
/root/Arduino/libraries/Mozzi/internal/MozziGuts.hpp:85:3: note: suggested alternative: 'AudioOutput'
   audioOutput(f);
   ^~~~~~~~~~~
   AudioOutput
/root/Arduino/libraries/Mozzi/internal/MozziGuts.hpp: In function 'void MozziPrivate::audioHook()':
/root/Arduino/libraries/Mozzi/internal/MozziGuts.hpp:232:7: error: 'canBufferAudioOutput' was not declared in this scope
   if (canBufferAudioOutput()) {
       ^~~~~~~~~~~~~~~~~~~~
/root/Arduino/libraries/Mozzi/internal/MozziGuts.hpp:232:7: note: suggested alternative: 'bufferAudioOutput'
   if (canBufferAudioOutput()) {
       ^~~~~~~~~~~~~~~~~~~~
       bufferAudioOutput
exit status 1
Erro compilando para a placa Arduino Nano RP2040 Connect

tomcombriat, 2024-09-20:
Hi,

Could you provide a bit more information? And especially which core are you using? Mozzi only works with this one for the RP2040: https://github.com/earlephilhower/arduino-pico

Other useful information could be to know if you are using the latest version of Mozzi.

tomcombriat, 2025-01-13:
@xxzykw Up

tomcombriat, 2025-01-14:
Closing because of lack of information and, as such, cannot reproduce. Happy to reopen if that was closed prematurely.

closed 2025-01-14 by tomcombriat