sensorium/Mozzi #278: The ESP32 S3 does not have a built-in DAC, but it has been identified as an ESP32 device and cannot use PWM analog output. Is there any way to solve the problem of using ESP32S3 normally without external or internal DAC

1183747933, 2024-10-05:
The ESP32 S3 does not have a built-in DAC, but it has been identified as an ESP32 device and cannot use PWM analog output. Is there any way to solve the problem of using ESP32S3 normally without external or internal DAC

tomcombriat, 2024-10-07:
Hi,
I never had an ESP32 in my hands so this is hypothetical but according to https://sensorium.github.io/Mozzi/#supported-boards-output-modes-and-default-pins the ESP32 default outputting mode is PDM on GPIO2. Have you tried that?

1183747933, 2024-10-07:
> Hi, I never had an ESP32 in my hands so this is hypothetical but according to https://sensorium.github.io/Mozzi/#supported-boards-output-modes-and-default-pins the ESP32 default outputting mode is PDM on GPIO2. Have you tried that?

Later, I tried to use the ESP32 to connect to an external DAC (PCM5102) to get sound, but it is not the S3. However, strangely enough, even when I don't perform any operations and simply include #include <Mozzi.h>, my ESP32 keeps rebooting. Below is my code.

-------------------------------------------------------------------------------------

```
//Mozzi library version 2.0.1, obtained from PlatformIO, using the chip module ESP32-WROOM-32.
// #include <Arduino.h>
// #include "MozziConfigValues.h" 
// #define MOZZI_AUDIO_MODE MOZZI_OUTPUT_I2S_DAC
// #define MOZZI_I2S_PIN_BCK  26
// #define MOZZI_I2S_PIN_WS   25
// #define MOZZI_I2S_PIN_DATA 22
// #define MOZZI_I2S_PORT I2S_NUM_AUTO

// #define MOZZI_I2S_PIN_BCK   14    //BCK
// #define MOZZI_I2S_PIN_WS   26   //LCK
// #define MOZZI_I2S_PIN_DATA   27   //DIN
// #define MOZZI_I2S_PORT I2S_NUM_AUTO

// #define MOZZI_CONTROL_RATE 64    // Hz, powers of 2 are most reliable; 64 Hz is actually the default, but shown here, for clarity
#include <Mozzi.h>  //! WARNING  REBOOT
// #include <Oscil.h> // oscillator template
// #include <tables/sin2048_int8.h> // sine table for oscillator
// Oscil <SIN2048_NUM_CELLS, MOZZI_AUDIO_RATE> aSin(SIN2048_DATA);

// void updateControl(){}

// AudioOutput_t updateAudio(){
//   return MonoOutput::from8Bit(aSin.next()); // return an int signal centred around 0
// }

void setup() {
  Serial.begin(115200);
  
  // startMozzi();
  // aSin.setFreq(440);

}

void loop() {
  // Serial.println(xPortGetCoreID());
  // audioHook();
  // delay(1000);
}

```

1183747933, 2024-10-07:
> > Hi, I never had an ESP32 in my hands so this is hypothetical but according to https://sensorium.github.io/Mozzi/#supported-boards-output-modes-and-default-pins the ESP32 default outputting mode is PDM on GPIO2. Have you tried that?
> 
> Later, I tried to use the ESP32 to connect to an external DAC (PCM5102) to get sound, but it is not the S3. However, strangely enough, even when I don't perform any operations and simply include #include <Mozzi.h>, my ESP32 keeps rebooting. Below is my code.-----------谷歌翻译-----------后来我尝试使用ESP32连接外部DAC（PCM5102）来获得声音，但它不是S3。然而，奇怪的是，即使我不执行任何操作并简单地包含 #include <Mozzi.h>，我的 ESP32 也会不断重新启动。下面是我的代码。
> 
> ```
> //Mozzi library version 2.0.1, obtained from PlatformIO, using the chip module ESP32-WROOM-32.
> // #include <Arduino.h>
> // #include "MozziConfigValues.h" 
> // #define MOZZI_AUDIO_MODE MOZZI_OUTPUT_I2S_DAC
> // #define MOZZI_I2S_PIN_BCK  26
> // #define MOZZI_I2S_PIN_WS   25
> // #define MOZZI_I2S_PIN_DATA 22
> // #define MOZZI_I2S_PORT I2S_NUM_AUTO
> 
> // #define MOZZI_I2S_PIN_BCK   14    //BCK
> // #define MOZZI_I2S_PIN_WS   26   //LCK
> // #define MOZZI_I2S_PIN_DATA   27   //DIN
> // #define MOZZI_I2S_PORT I2S_NUM_AUTO
> 
> // #define MOZZI_CONTROL_RATE 64    // Hz, powers of 2 are most reliable; 64 Hz is actually the default, but shown here, for clarity
> #include <Mozzi.h>  //! WARNING  REBOOT
> // #include <Oscil.h> // oscillator template
> // #include <tables/sin2048_int8.h> // sine table for oscillator
> // Oscil <SIN2048_NUM_CELLS, MOZZI_AUDIO_RATE> aSin(SIN2048_DATA);
> 
> // void updateControl(){}
> 
> // AudioOutput_t updateAudio(){
> //   return MonoOutput::from8Bit(aSin.next()); // return an int signal centred around 0
> // }
> 
> void setup() {
>   Serial.begin(115200);
>   
>   // startMozzi();
>   // aSin.setFreq(440);
> 
> }
> 
> void loop() {
>   // Serial.println(xPortGetCoreID());
>   // audioHook();
>   // delay(1000);
> }
> ```

The information returned by the serial is as follows:
```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:13228
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:2924
entry 0x40080570

abort() was called at PC 0x400d2c9f on core 0

Backtrace: 0x40082369:0x3ffe3b50 0x400866c1:0x3ffe3b70 0x40089429:0x3ffe3b90 0x400d2c9f:0x3ffe3c00 0x400d4902:0x3ffe3c20 0x400818ea:0x3ffe3c50 0x40079312:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: a37a9806384b2877

Rebooting...
```

tomcombriat, 2024-10-08:
Additionally, have you tried narrowing down the reboot reason by using for instance the technique described here: https://forum.arduino.cc/t/esp32-esp-wroom-32-and-wdt/1005179/4 ?

tfry-git, 2024-10-10:
This is probably the same issue as https://github.com/sensorium/Mozzi/issues/261 .

1183747933, 2024-10-23:
> Additionally, have you tried narrowing down the reboot reason by using for instance the technique described here: https://forum.arduino.cc/t/esp32-esp-wroom-32-and-wdt/1005179/4 ?

I'm sorry, due to a job transfer, I don't have time to experiment with this issue at the moment. Thank you very much for your reply. I will conduct another test in a while.

closed 2025-02-24 by tomcombriat