sensorium/Mozzi #279: 8-bit Sine tables to not describe sines, but rather negative cosines

eclab, 2024-10-10:
Most of the sine tables do not start at the origin, but rather at -128.  This is bad because it can cause a click if the wave is played raw.  They should start at 0, then increase from there, then back down to -128.

tomcombriat, 2024-10-11:
Hi, 
I am not sure I completely understand the issue. Looking at the sine tables, only `sin1024_int8.h` starts at -128, the others correctly start at 0. I agree that the stated types of the tables are not always coherent (and `tables/sin1024_uint8.h` looks a bit weird with a lot of padding). A bit of cleanup might be needed.

Or are you talking about the cos tables? In that case, the `cosphase` are also of interest.

eclab, 2024-10-11:
sin1024_int8.h
sintest_int8.h
sin1024_uint8.t
sin8192_uint8.t

These four start at the wrong values.  The uint ones should be starting at 128 (that's their origin).

I use sin1024 heavily for 8-bit sounds and this discovery was unfortunate.

tomcombriat, 2024-10-11:
Okay, thanks for the clarification. Here is my take on these tables:

 - sin1024_int8 **should be fixed** I agree, hopefully no-one is using the wrong behavior. I quite don't understand how that bring some click as the error seems only to be on the phase, hence the wave is still continuous, just not starting at the correct spot. I have not plotted it though.
 - sintest **should be deleted**. All its content is commented anyway
 - sin1024_uint8 and sin8192_uint are, IMO, more confusing than anything else, their content is not matching the said type, and having a trigonometric function which is not actually 0-centered sounds a bit off. I would be **in favor of removing them** (or deprecating them).

I am interested to have the opinions of @tfry-git and @sensorium before further action though.

eclab, 2024-10-11:
sin1024_int8 creates a click when you start the wave in an Oscil because it's not starting at 0 so it pops from 0 to -128 for its first sample output.  As to the uint variants, I presumed they're for unipolar modulation.

PR #284 (2024-11-12): Fix/tables
Fix #279
closed 2024-11-12 by tomcombriat