sensorium/Mozzi #288: New release?

tomcombriat, 2024-11-28:
Hello!

I regenerated the docs (and fixed the logo) and pushed that straight to the main repo, hope that is okay with you.

I drafted a new release: https://github.com/sensorium/Mozzi/releases if no-one is against I will publish it in the next few days!

Best,
Tom

sensorium, 2024-11-28:
Hey Tom,
one thing I noticed when installing Mozzi recently was that in the Arduino library manager, the credits presently don't include you and @tfry-git and others.  Maybe that could be changed from my name to "Mozzi Team" or something?

tomcombriat, 2024-11-29:
Note that currently it writes:

author=Tim Barrass and contributors as documented in source, and at https://github.com/sensorium/Mozzi/graphs/contributors

which sound reasonable?

closed 2025-03-30 by tomcombriat