sensorium/Mozzi #290: Config for Sparkfun Pro Mini esp32?

sensorium, 2024-11-28:
I can't get audio from the Sparkfun Pro Mini esp32.  It worked 6 months ago (before Mozzi 2.0?).  I don't remember any special config back then, and there's nothing in the sketches.  But I expect the issue is with my brain rather than the software....

Should this work, at the top of the sketch?
#include "MozziConfigValues.h"
#define MOZZI_AUDIO_MODE MOZZI_OUTPUT_INTERNAL_DAC
#define MOZZI_AUDIO_CHANNELS MOZZI_MONO

tomcombriat, 2024-11-29:
Hi @sensorium 

I'm not completely sure it is your brain, there was a few reports of problems with ESP32 lately (#278 #261 ). Unfortunately I do not have the hardware yet to reproduce.

One thing is sure (even though the main webpage is wrong) is that the default output mode for ESP32 should be the internal dac (https://github.com/sensorium/Mozzi/blob/091c31998b11d32c68483db4b9c9b96fef2b5550/internal/config_checks_esp32.h#L86) so normally no config should be needed… You do not get any sound on any pin? No warning or errors on compilation? Have you tried with Mozzi1?

sensorium, 2024-12-11:
I'll close this and open a new issue as it seems just including Mozzi.h stops normal operation of the board.

closed 2024-12-11 by sensorium