sensorium/Mozzi #292: On-the-fly custom wavetable

baritonomarchetto, 2024-12-04:
Hi folks, I have breadborded a Pi Pico with a TFT screen with touch. My goal is to draw a waveform on the screen and have it outputted as audio.

My question is: is it possible to actually load wavetables on the fly? If yes, what is the format (wavetables coming with Mozzi look like simple arrays to me) or "trick" to do so?

Thanks

sensorium, 2024-12-04:
Hi, here's an old sketch which seems to work, changing a table on the fly.  It has a custom Oscil class to use a wavetable in update-able ram.
[realtimeTablewrite.zip](https://github.com/user-attachments/files/18015127/realtimeTablewrite.zip)

tomcombriat, 2024-12-05:
Hi,
I am a bit on thin ice because I have not tested but as `Oscil` expect a `const int8_t *` in `setTable()` and that type can be upcastted from a `int8_t *` (I think, that's what I am not completely sure, from what I understand the `const` in this case is a guarantee that the function will not modify the content pointed by the pointer, not that this content should be constant everywhere), it should also work with a stock `Oscil`?

baritonomarchetto, 2024-12-05:
> Hi, here's an old sketch which seems to work, changing a table on the fly. It has a custom Oscil class to use a wavetable in update-able ram. [realtimeTablewrite.zip](https://github.com/user-attachments/files/18015127/realtimeTablewrite.zip)

I made a try porting the sketch to Mozzi V2, but it doesn't work. Maybe I made something wrong, maybe it's no more compatible. I only touched "realtimeTableWrite.ino", not "WriteableOscil.h".
Could you please take a look?

`/*  Example modifying a wavetable on the fly
    using Mozzi sonification library.

    Demonstrates the use of WriteableOscil to play and modify a wavetable.
*/

#define MOZZI_CONTROL_RATE 64 // Hz, powers of 2 are most reliable
#include <Mozzi.h>
#include "WriteableOscil.h" // local oscillator template

int8_t wave_data []  =
        {
                -1, 3, 6, 9, 12, 15, 18, 21, 24, 28,
                31, 34, 37, 40, 43, 46, 48, 51, 54, 57, 60, 63, 65, 68, 71, 73, 76, 78, 81, 83,
                85, 88, 90, 92, 94, 96, 98, 100, 102, 104, 106, 108, 109, 111, 112, 114, 115,
                117, 118, 119, 120, 121, 122, 123, 124, 124, 125, 126, 126, 127, 127, 127, 127,
                127, 127, 127, 127, 127, 127, 127, 126, 126, 125, 124, 124, 123, 122, 121, 120,
                119, 118, 117, 115, 114, 112, 111, 109, 108, 106, 104, 102, 100, 98, 96, 94, 92,
                90, 88, 85, 83, 81, 78, 76, 73, 71, 68, 65, 63, 60, 57, 54, 51, 48, 46, 43, 40,
                37, 34, 31, 28, 24, 21, 18, 15, 12, 9, 6, 3, 0, -4, -7, -10, -13, -16, -19, -22,
                -25, -29, -32, -35, -38, -41, -44, -47, -49, -52, -55, -58, -61, -64, -66, -69,
                -72, -74, -77, -79, -82, -84, -86, -89, -91, -93, -95, -97, -99, -101, -103,
                -105, -107, -109, -110, -112, -113, -115, -116, -118, -119, -120, -121, -122,
                -123, -124, -125, -125, -126, -127, -127, -128, -128, -128, -128, -128, -128,
                -128, -128, -128, -128, -128, -127, -127, -126, -125, -125, -124, -123, -122,
                -121, -120, -119, -118, -116, -115, -113, -112, -110, -109, -107, -105, -103,
                -101, -99, -97, -95, -93, -91, -89, -86, -84, -82, -79, -77, -74, -72, -69, -66,
                -64, -61, -58, -55, -52, -49, -47, -44, -41, -38, -35, -32, -29, -25, -22, -19,
                -16, -13, -10, -7, -4,
        };
        
// use: Oscil <table_size, update_rate> oscilName (wavetable), look in .h file of table #included above
WriteableOscil <256, MOZZI_AUDIO_RATE> aWave(wave_data);

void setup(){
  startMozzi();
  aWave.setFreq(400); // set the frequency
}

void updateControl(){
  byte cell_to_change = random(256);
  int cell_value = wave_data[cell_to_change];
  cell_value = cell_value+(random(2)-1); //+1 or -1
  // bounce at wave range limits
  if(cell_value>127) cell_value = cell_value-2;
  if(cell_value<-127) cell_value = cell_value+2;
  wave_data[cell_to_change]  = cell_value;
}

AudioOutput updateAudio(){
  return MonoOutput::from8Bit(aWave.next()); // return an int signal centred around 0
}

void loop(){
  audioHook(); // required here
}`

sensorium, 2024-12-05:
@baritonomarchetto, that's weird, I tried it today with up-to-date Mozzi and it worked without any modification... _ah, but that was on an avr..._

@tomcombriat knows about other platforms.

tomcombriat, 2024-12-05:
@sensorium Did you try on an AVR? The ATmega168 is from the same family.

@baritonomarchetto I'll try to give it a try later today, if you have a pico lying around might be worth trying on that? Also, did you confirm that your hardware configuration is correct by successfully having one of the basic example to output sound?

baritonomarchetto, 2024-12-05:
Ok, on nano ATmega168 the code I updated works, but sounds weird (crackling and unpleasant). There must be still something wrong here.

About the Pi Pico: the same code (again, the one I uploaded) returns an error while compiling. I am using Arduino IDE with Earle Phil Hower III pi pico core.
Here is the error:
(thank you very much for the help so far!!)

```
FQBN: rp2040:rp2040:rpipico
Using board 'rpipico' from platform in folder: C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1
Using core 'rp2040' from platform in folder: C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1

Detecting libraries used...
C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\4.0.1-8ec9d6f/bin/arm-none-eabi-g++ -I C:\Users\HOME\AppData\Local\Temp\arduino\sketches\6A5BE6FF6D49842A14C44AA452C8DDEE/core -c -Werror=return-type -Wno-psabi -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico" -DPICO_FLASH_SIZE_BYTES=2097152 @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/lib/rp2040/platform_def.txt -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefixC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/ @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/lib/rp2040/platform_inc.txt @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/lib/core_inc.txt -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO -DBOARD_NAME="RASPBERRY_PI_PICO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1\cores\rp2040 -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1\variants\rpipico C:\Users\HOME\AppData\Local\Temp\arduino\sketches\6A5BE6FF6D49842A14C44AA452C8DDEE\sketch\Graphical_Wavetable_VCO.ino.cpp -o nul
Alternatives for MozziConfigValues.h: [Mozzi@2.0.1]
ResolveLibrary(MozziConfigValues.h)
  -> candidates: [Mozzi@2.0.1]
C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\4.0.1-8ec9d6f/bin/arm-none-eabi-g++ -I C:\Users\HOME\AppData\Local\Temp\arduino\sketches\6A5BE6FF6D49842A14C44AA452C8DDEE/core -c -Werror=return-type -Wno-psabi -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico" -DPICO_FLASH_SIZE_BYTES=2097152 @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/lib/rp2040/platform_def.txt -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefixC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/ @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/lib/rp2040/platform_inc.txt @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/lib/core_inc.txt -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO -DBOARD_NAME="RASPBERRY_PI_PICO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1\cores\rp2040 -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1\variants\rpipico -Ic:\Users\HOME\Documents\Arduino\libraries\Mozzi C:\Users\HOME\AppData\Local\Temp\arduino\sketches\6A5BE6FF6D49842A14C44AA452C8DDEE\sketch\Graphical_Wavetable_VCO.ino.cpp -o nul
Alternatives for FixMath.h: [FixMath@1.0.6]
ResolveLibrary(FixMath.h)
  -> candidates: [FixMath@1.0.6]
C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\4.0.1-8ec9d6f/bin/arm-none-eabi-g++ -I C:\Users\HOME\AppData\Local\Temp\arduino\sketches\6A5BE6FF6D49842A14C44AA452C8DDEE/core -c -Werror=return-type -Wno-psabi -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico" -DPICO_FLASH_SIZE_BYTES=2097152 @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/lib/rp2040/platform_def.txt -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -iprefixC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/ @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/lib/rp2040/platform_inc.txt @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/lib/core_inc.txt -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO -DBOARD_NAME="RASPBERRY_PI_PICO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1\cores\rp2040 -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1\variant [... cut ...]
Using library Wire at version 1.0 in folder: C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1\libraries\Wire 
Using library SPI at version 1.0 in folder: C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1\libraries\SPI 
Using library Adafruit ST7735 and ST7789 Library at version 1.10.4 in folder: C:\Users\HOME\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library 
exit status 1

Compilation error: invalid conversion from 'const int8_t*' {aka 'const signed char*'} to 'int8_t*' {aka 'signed char*'} [-fpermissive]

```

tomcombriat, 2024-12-05:
Ah yes, all these toolchains are on different C++ standards…

I'll to try myself, but first you can try removing the `const`  from l.72 and 100 of WritableOscil. (if it still complains you can remove all const of that file…). I kind of agree with the compiler on this one: the function promises it won't modify the content of the table with `const` but cast it to non-const, hence have the potential to break this promise. A stock `Oscil` should be more coherent there (if the upcast to `const` is accepted by the compiler). More information once I get behind a breadboard ;)

baritonomarchetto, 2024-12-05:
By removing those two "const" makes the code compile, even on Pi Pico.

The bad-sounding thing still remains, both on nano (even worse here) and Pi Pico. 
If I am correct, I should at least hear a wave morphing with time. The only thing I hear is a fixed, bad sounding (very bad on nano) wave :)

(many thanks for the help so far)

sensorium, 2024-12-05:
Mmm so what you sent sounded right on my pro mini 328 board, a sine wave slowly degrading as the wavetable gets more randomised... 
Is the clock speed of your atmega168 16MHz?  I've never tried one and not sure what's different to 328s.

Also, I made an error in the randomising, so here's that fixed (I think) and the degrading made more noticeable:
(You can probably think of a better way to test changing the table).

        void updateControl(){
          byte cell_to_change = random(256);
          int cell_value = wave_data[cell_to_change];
          cell_value = cell_value+(random(7)-3); //-3 -> +3
          // bounce at wave range limits
          if(cell_value>127) cell_value = cell_value-5;
          if(cell_value<-127) cell_value = cell_value+5;
          wave_data[cell_to_change]  = cell_value;
        }

baritonomarchetto, 2024-12-05:
Looks like the nano 168 was the issue, maybe as you said the clock (8MHz).

Tested with an arduino UNO and it sounds as it should.

On Pi Pico, I can heard a steady sinewave (a little bit degraded in my opinion) , but this is because randomseeding is not supported there.

Attached are the files, updated with my update and fixed randomseeding, for future reference. I also added a little description of what one should heard.

This issue can be labelled as closed. I will likely open some other ticket for help in my current project.

Thanks!!

[realtimeTablewrite.zip](https://github.com/user-attachments/files/18024041/realtimeTablewrite.zip)

tomcombriat, 2024-12-05:
Nice!

Out of curiosity, have you also tried with a stock `Oscil`?

Even if the pico does not have random seeding (I actually now remember someone sending a PR on that…) the sound should also evolve right? Just exactly in the same way for each run…

tomcombriat, 2024-12-05:
> but doesn't degrade with time because of missing randomseeding

It should (and it does according to my testings). The fact that there is no random seed means that the numbers returned by `random` will be the same between every reboot, but it still returns a pseudo-random sequence.

That's nice, we should add an example on that!

PS: I am working on similar hardware (Pico + TFT + DAC). a bit of a mess as of now but might be useful.
https://github.com/tomcombriat/G_TES_touch-synth
https://github.com/tomcombriat/GT_TouchSynth_lib

Closing as completed.

closed 2024-12-05 by tomcombriat