sensorium/Mozzi #296: Raspberry Pi Pico audio bits and output resolution sanity chech

baritonomarchetto, 2024-12-09:
Hello everybody, not really an issue but a clarification.

I am working on a wavetable oscillator module based on Pi Pico microcontroller board and Mozzi library.

I am using the default MOZZI_OUTPUT_PWM mode. By reading the official [documentation  on RP2040](https://sensorium.github.io/Mozzi/doc/html/hardware_rp2040.html) the default audio bits is set to 11 (MOZZI_AUDIO_BITS 11).

Now, what should be the "correct" output frame I should use? MonoOutput::from**8**Bit(etc. etc.) or MonoOutput::fromNBit(**11**, etc. etc.)? I tried both and my impression is that using fromNbit lowens the volume and nothing else.

The tables I am using are those included with Mozzi (i.e sin2048_int8.h) which are 8-bit.

Even if I read the documentation and took a look to most of the relevant examples, I still make some confusion between audio bits, wavetable bits and PWM resolution...

Thanks in advance for any help

tomcombriat, 2024-12-09:
Hi,

It actually depends on what you are putting in `MonoOutput::fromNBits`, the whole idea is that the fact that the pico is outputting 11bits should not come into play. This allows for the same sketches to be used on different platforms, outputting different numbers of bits.

The `N` in `MonoOutput::fromNBits(N, sig)` is the number of bits of `sig`, not the number of bits of your platform. Mozzi will automatically convert from these N bits to the actual number of bits needed.

For instance, if you only have an `Oscil aSin`, as this will output 8bits you should use `MonoOutput::fromNBits(8, aSin.next())` (or the 8bits version, that's the same in the end).

If you have two say (pseudo-code, won't compile):
```
Oscil aSin1;
Oscil aSin2;
```
then both `MonoOutput::fromNBits(9, aSin1.next() + aSin2.next())` and  `MonoOutput::fromNBits(16, aSin1.next() * aSin2.next())` are correct (but the sound won't be the same of course).

baritonomarchetto, 2024-12-09:
Ah, now it's more clear!

When you say "the actual number of bits needed" you are referring to MOZZI_AUDIO_BITS, which is 11 by default on Pico, right?

tomcombriat, 2024-12-09:
> When you say "the actual number of bits needed" you are referring to MOZZI_AUDIO_BITS, which is 11 by default on Pico, right?

Yes. But the main idea is that you can forget how many bits your platform is outputting, that's Mozzi's job ;). You just have to tell Mozzi how many bits you are feeding in. This is why different examples have different `N` but still sounds okay with no volume change.

closed 2024-12-09 by baritonomarchetto