sensorium/Mozzi #298: PT8211_stereo_16bits not working on PI PICO?

baritonomarchetto, 2024-12-14:
Hello Team,

I wanted to give PT8211 DAC a try on my Pi Pico. Uploaded the default sketch (I only changed the channel select pin to GPIO 11) but it doesnt work.

Could anyone confirm please?

Thanks!

tomcombriat, 2024-12-14:
This is strange, I am fairly sure I got that config to work (and uses the mono equivalent fairly often).

> I only changed the channel select pin to GPIO 11)

Is that the pin linked to `WS`? This is **hardcoded** (not by Mozzi but by the arduino-pico port of I2S) to being the one above the clock pin:
```
 //#define MOZZI_IS_PIN_WS(MOZZI_I2S_PIN_BCK + 1)  // implicit
```

If that still does not work, could you provide a minimal non working example (especially the config variables on top)? And maybe detail exactly your pinout?

baritonomarchetto, 2024-12-15:
Thanks. By usign default SPI (MISO = GPIO 16, MOSI = GPIO 19, CLK = GPIO 18) shouldn't WS be pin 19 (the one above CLK)? But pin 19 is already assigned to MOSI. What am I missing?

tomcombriat, 2024-12-15:
Hi,
I think I now understand where the confusion stemmed from! SPI and I2S are two different protocols (even though there are fairly similar). On some platforms which do not have I2S, it is possible to emulate it using SPI. However, the Pico does have [I2S](https://arduino-pico.readthedocs.io/en/latest/i2s.html#i2s-output), and with fully configurable pins, so you do not have to use SPI ready pin to use it.

Does it work?

baritonomarchetto, 2024-12-15:
No, it does not. I am not sure how the audioOutput function should be. This is my actual, not working but compiling full code (an attempt to have the PT8211 example working with I2S, so it's someway a mixed beast. I do not master comunication protocols, unfortunately).

`// before including Mozzi.h, configure external audio output mode:
#include "MozziConfigValues.h"  // for named option values
#define MOZZI_AUDIO_MODE MOZZI_OUTPUT_EXTERNAL_TIMED
#define MOZZI_AUDIO_CHANNELS MOZZI_STEREO
#define MOZZI_CONTROL_RATE 256 // Hz, powers of 2 are most reliable
#define MOZZI_I2S_PIN_BCK 20 //default. To DAC pin 1 (BCK)
#define MOZZI_I2S_PIN_WS (MOZZI_I2S_PIN_BCK+1) // CANNOT BE CHANGED, HAS TO BE NEXT TO pBCLK, i.e. default is 21
#define MOZZI_I2S_PIN_DATA 22 //default. To DAC pin 3 (DIN)
#define MOZZI_I2S_FORMAT MOZZI_I2S_FORMAT_PLAIN //default

#include <Mozzi.h>
#include <Oscil.h>
#include <tables/cos2048_int8.h> // table for Oscils to play
#include <SPI.h>

// Synthesis part
Oscil<COS2048_NUM_CELLS, MOZZI_AUDIO_RATE> aCos1(COS2048_DATA);
Oscil<COS2048_NUM_CELLS, MOZZI_AUDIO_RATE> aCos2(COS2048_DATA);
Oscil<COS2048_NUM_CELLS, MOZZI_CONTROL_RATE> kEnv1(COS2048_DATA);
Oscil<COS2048_NUM_CELLS, MOZZI_CONTROL_RATE> kEnv2(COS2048_DATA);

// External audio output parameters
#define MOZZI_I2S_PIN_WS 21   // channel select pin for the DAC. This is hardcoded and cannot be changed

void audioOutput(const AudioOutput f){ // f is a structure containing both channels
/* Note:
 *  the digital writes here can be optimised using portWrite if more speed is needed
 */
  digitalWrite(MOZZI_I2S_PIN_WS, LOW);  //select Right channel
  SPI.transfer16(f.r());      // Note: This DAC works on 0-centered samples, no need to add MOZZI_AUDIO_BIAS

  digitalWrite(MOZZI_I2S_PIN_WS, HIGH);  // select Left channel
  SPI.transfer16(f.l());
}

void setup() {
  pinMode(MOZZI_I2S_PIN_WS, OUTPUT);
  // Initialising the SPI connection on default port
  SPI.begin();
  SPI.beginTransaction(SPISettings(20000000, MSBFIRST, SPI_MODE0));  //MSB first, according to the DAC spec
  aCos1.setFreq(440.f);
  aCos2.setFreq(220.f);
  kEnv1.setFreq(0.25f);
  kEnv2.setFreq(0.30f);
  startMozzi();
}

// Carry enveloppes
int env1, env2;

void updateControl() {
  env1 = kEnv1.next();
  env2 = kEnv2.next();
}

AudioOutput updateAudio() {
  return StereoOutput::from16Bit(aCos1.next() * env1, aCos2.next() * env2);
}

void loop() {
  audioHook();
}`

tomcombriat, 2024-12-15:
I2S is a *native* output mode mode for the RP2040/Mozzi. This means you do not even have to define an `AudioOutput()` function (which is only for *custom* ways of outputting the sound).

So your code should work if you remove the `void audioOutput(const AudioOutput f)` block.
You can also remove `#define MOZZI_I2S_PIN_WS (MOZZI_I2S_PIN_BCK+1) // CANNOT BE CHANGED, HAS TO BE NEXT TO pBCLK, i.e. default is 21` from your config (this is done by Mozzi) and `#include <I2S.h>` (same reason).

`void audioOutput(const AudioOutput f)` is only to be used when the Mozzi's output mode is set to `MOZZI_OUTPUT_EXTERNAL_TIMED` or  `MOZZI_OUTPUT_EXTERNAL_CUSTOM` (see https://sensorium.github.io/Mozzi/doc/html/group__audio__output.html and maybe https://sensorium.github.io/Mozzi/doc/html/group__config.html#config_general (look for `MOZZI_AUDIO_MODE`).

baritonomarchetto, 2024-12-15:
Removed the function, now I have what looks like to be an issue with SPI. The error IDE returns is `Compilation error: expected unqualified-id before '.' token` referring to `SPI.begin();`.

This is the full, updated code for reference (thank you very much Tom!)

```
// before including Mozzi.h, configure external audio output mode:
#include "MozziConfigValues.h"  // for named option values
#define MOZZI_AUDIO_MODE MOZZI_OUTPUT_I2S_DAC
#define MOZZI_AUDIO_CHANNELS MOZZI_STEREO
#define MOZZI_CONTROL_RATE 256 // Hz, powers of 2 are most reliable
#define MOZZI_I2S_PIN_BCK 20 //default. To DAC pin 1 (BCK)
//#define MOZZI_I2S_PIN_WS (MOZZI_I2S_PIN_BCK+1) // CANNOT BE CHANGED, HAS TO BE NEXT TO pBCLK, i.e. default is 21. To DAC pin 2 (WS)
#define MOZZI_I2S_PIN_DATA 22 //default. To DAC pin 3 (DIN)
#define MOZZI_I2S_FORMAT MOZZI_I2S_FORMAT_LSBJ //PT8211 mode as per datasheet

#include <Mozzi.h>
#include <Oscil.h>
#include <tables/cos2048_int8.h> // table for Oscils to play
//#include <I2S.h> already included

// Synthesis part
Oscil<COS2048_NUM_CELLS, MOZZI_AUDIO_RATE> aCos1(COS2048_DATA);
Oscil<COS2048_NUM_CELLS, MOZZI_AUDIO_RATE> aCos2(COS2048_DATA);
Oscil<COS2048_NUM_CELLS, MOZZI_CONTROL_RATE> kEnv1(COS2048_DATA);
Oscil<COS2048_NUM_CELLS, MOZZI_CONTROL_RATE> kEnv2(COS2048_DATA);

// External audio output parameters
#define MOZZI_I2S_PIN_WS 21   // channel select pin for the DAC. This is hardcoded and cannot be changed

void setup() {
  pinMode(MOZZI_I2S_PIN_WS, OUTPUT);
  I2S.begin();
  // Initialising the SPI connection on default port
  //SPI.begin();
  //SPI.beginTransaction(SPISettings(20000000, MSBFIRST, SPI_MODE0));  //MSB first, according to the DAC spec
  aCos1.setFreq(440.f);
  aCos2.setFreq(220.f);
  kEnv1.setFreq(0.25f);
  kEnv2.setFreq(0.30f);
  startMozzi();
}

// Carry eveloppes
int env1, env2;

void updateControl() {
  env1 = kEnv1.next();
  env2 = kEnv2.next();
}

AudioOutput updateAudio() {
  return StereoOutput::from16Bit(aCos1.next() * env1, aCos2.next() * env2);
}

void loop() {
  audioHook();
}
```

tomcombriat, 2024-12-15:
Remove the I2S.begin() in your setup I think. This is also mozzi's job;).

Normally, you only have to set the options (all the define on top), and no further actions is required from your part. (The examples should work as is, except for the options).

baritonomarchetto, 2024-12-15:
Removing `I2S.begin()` makes it compile, but I have nothing at the output.

Everithing is wired correctly...

GPIO 20       -> PT8211 pin 1 (BCK)
GPIO 21       -> PT8211 pin 2 (WS)
GPIO 22       -> PT8211 pin 3 (DIN)
GND            -> PT8211 pin 4 (GND)
VCC             -> PT8211 pin 5 (3.3V)
L_CH_OUT   -> PT8211 pin 6 (3.3V)
n.c               -> PT8211 pin 7 
R_CH_OUT  -> PT8211 pin 8 (3.3V)

The current code (notice that I have removed completely the WS definition: I suppose it is set to `output` by Mozzi):

```
// before including Mozzi.h, configure external audio output mode:
#include "MozziConfigValues.h"  // for named option values
#define MOZZI_AUDIO_MODE MOZZI_OUTPUT_I2S_DAC
#define MOZZI_AUDIO_CHANNELS MOZZI_STEREO
#define MOZZI_CONTROL_RATE 256 // Hz, powers of 2 are most reliable
#define MOZZI_I2S_PIN_BCK 20 //default. To DAC pin 1 (BCK)
//#define MOZZI_I2S_PIN_WS (MOZZI_I2S_PIN_BCK+1) // CANNOT BE CHANGED, HAS TO BE NEXT TO pBCLK, i.e. default is 21. To DAC pin 2 (WS)
#define MOZZI_I2S_PIN_DATA 22 //default. To DAC pin 3 (DIN)
#define MOZZI_I2S_FORMAT MOZZI_I2S_FORMAT_LSBJ //PT8211 mode as per datasheet

#include <Mozzi.h>
#include <Oscil.h>
#include <tables/cos2048_int8.h> // table for Oscils to play
//#include <I2S.h> already included

// Synthesis part
Oscil<COS2048_NUM_CELLS, MOZZI_AUDIO_RATE> aCos1(COS2048_DATA);
Oscil<COS2048_NUM_CELLS, MOZZI_AUDIO_RATE> aCos2(COS2048_DATA);
Oscil<COS2048_NUM_CELLS, MOZZI_CONTROL_RATE> kEnv1(COS2048_DATA);
Oscil<COS2048_NUM_CELLS, MOZZI_CONTROL_RATE> kEnv2(COS2048_DATA);

void setup() {
  // Initialising I2S connection
  //I2S.begin(); already included
  aCos1.setFreq(440.f);
  aCos2.setFreq(220.f);
  kEnv1.setFreq(0.25f);
  kEnv2.setFreq(0.30f);
  startMozzi();
}

// Carry envelopes
int env1, env2;

void updateControl() {
  env1 = kEnv1.next();
  env2 = kEnv2.next();
}

AudioOutput updateAudio() {
  return StereoOutput::from16Bit(aCos1.next() * env1, aCos2.next() * env2);
}

void loop() {
  audioHook();
}
```
Thank you very much for the help so far!

tomcombriat, 2024-12-15:
Slowly getting there ^^.

I can test here a bit later, but maybe first, let's simplify!

You're using the default pins, so maybe something like that (very simple sketch, mono) should work out of the bat:

``` c++
#include "MozziConfigValues.h"  // for named option values
#define MOZZI_AUDIO_MODE MOZZI_OUTPUT_I2S_DAC
#define MOZZI_I2S_FORMAT MOZZI_I2S_FORMAT_LSBJ
#define MOZZI_AUDIO_CHANNELS 1

#include <Mozzi.h>
#include <Oscil.h> // oscillator template
#include <tables/sin2048_int8.h> // sine table for oscillator

// use: Oscil <table_size, update_rate> oscilName (wavetable), look in .h file of table #included above
Oscil <SIN2048_NUM_CELLS, MOZZI_AUDIO_RATE> aSin(SIN2048_DATA);

void setup(){
  startMozzi(); // :)
  aSin.setFreq(440); // set the frequency
}
void updateControl(){
  // put changing controls in here
}

AudioOutput updateAudio(){
  return MonoOutput::from8Bit(aSin.next()); // return an int signal centred around 0
}

void loop(){
  audioHook(); // required here
}
```

If you have a scope at hand, maybe worth checking the different signals that are sent to the DAC…

baritonomarchetto, 2024-12-15:
(many thanks)

Uploaded your sketch, I have no audio. The scope tells me something is sent to the DAC, on all three lines (clock, WS and DIN). No output though.
On a side note: I am using Earle Phil Hower Pico core. Could it be the culript?

baritonomarchetto, 2024-12-15:
I am on the latest Pico core (4.3.1). 

I bought DACs from AliExpress, so likely not legit. In total honesty, I bought other DACs there in the past with no issues.

It's the first time I use these PT8211, so I have never heard them work. I have actually never used PT8211 before and bought them to test your examples and see if they could boost Mozzi output quality. I am working on a wavetable oscillator eurorack module for my DIY modular synthesizer, actually adopting PWM on Pico but not sounding good enought at the moment :) 

Sure I can share a pic of my breadboard :)

[image]

baritonomarchetto, 2024-12-15:
Ok, your mono sketch works! I was expecting an AC signal, so I set the scope for AC. sorry. 

I can now see a DC biased (a little less than 1V) sine wave with 2V of amplitude, more or less.

With the scope I can also see my previous sketch working!

Not sure why I cannot hear anything on headphones: as soon as I connect my hearphones, signal goes straight to 400 mV and stays there. I had no issue with the connector I am using and PWM...

tomcombriat, 2024-12-15:
Nice!!

> so I set the scope for AC

Normally you should have seen something there too, but I guess it depends how high the scope filters out the low frequencies… 

You can easily turn that signal to pure AC with decoupling cap if needed.

> Does DAC signal need amplification to be heard?

Could not find the output current on the datasheet, but that really sounds that the DAC cannot output enough current to drive headphones (these are fairly low impedance…). I would not be surprised at all that PWM are more beefy.

> In your experience, does PT8211 signal need amplification to be heard?

I actually just tried. I can hear something with some 55ohms headphones, not that loud though… I (nearly) always filter and adapt the impedance before outputting, or I input into a sound card/amp which have quite high impedance. As of now, I have landed on [this](https://sensorium.github.io/Mozzi/learn/output_circuits/#more-advanced-filter-with-volume-control), but I sometimes started with just a buffer (beware that a lot of op-amp are not rail-to-rail!)

baritonomarchetto, 2024-12-15:
Thank you very much Tom for the help and informative (and enjoyable) chat. You helped me a lot :)

I will likely open other "issues" here, but if there's a better place to post for help requests instead of increasing Mozzi's issue count even when issue is not, please let me know

closed 2024-12-15 by baritonomarchetto

tomcombriat, 2024-12-15:
> Thank you very much Tom for the help and informative (and enjoyable) chat. You helped me a lot :)

Very welcome!

>  if there's a better place to post for help requests instead of increasing Mozzi's issue count even when issue is not, please let me know

Hum, there is always the chat where a bit more people are, but I am fairly less reactive there (and another maintainer is not on it). I think this one belonged here as it is not "How can I make this sound?" or something like that, but more a potential problem with Mozzi itself. Also, at 50-ish issues (some long overdue) I don't think it clutters that much, but that's only me ;).

So I guess, it depends a bit on the question :)!