sensorium/Mozzi #299: Pi Pico, MCP4728 AND Mozzi compatibility

baritonomarchetto, 2024-12-20:
Hello Team,

I am trying to use a MCP4728 I2C quad DAC with Mozzi for a 4 channels project (I know, actually Mozzi support 2 channel max, but I am trying to face one problem at a time :) ).

This is a basic sketch where I try to send a sinewave to the DAC. The sketch compiles, but after I upload it to my Pi Pico the microcontroller board get stuck (it is not even more possible to upload a new sketch and I need to go to UF2 mode to unbrick the processor).

The DAC works if I upload the example sketch that comes with the library (Benoit Schilling's MPC4728), so DAC library and wirings are ok.

Is there something obvious I am doing wrong softwarewise?

```
#define MOZZI_AUDIO_MODE MOZZI_OUTPUT_EXTERNAL_TIMED
#define MOZZI_CONTROL_RATE 64    // Hz, powers of 2 are most reliable; 64 Hz is actually the default, but shown here, for clarity
#include <Mozzi.h>
#include <Oscil.h> // oscillator template
#include <tables/sin2048_int8.h> // sine table for oscillator

#include <Wire.h>
#include "mcp4728.h"

// use: Oscil <table_size, update_rate> oscilName (wavetable), look in .h file of table #included above
Oscil <SIN2048_NUM_CELLS, MOZZI_AUDIO_RATE> aSin(SIN2048_DATA);

mcp4728 dac = mcp4728(0); // instantiate mcp4728 object, Device ID = 0

const int LDACpin = 2;
//SDA_pin = 0; //SDA0, default
//SCL0_pin = 1;//SCL0, default

void setup(){
  startMozzi(); // :)
  aSin.setFreq(440); // set the frequency

  dac.begin();  // initialize i2c interface
  //dac.vdd(5000); // set VDD(mV) of MCP4728 for correct conversion between LSB and Vout
  pinMode(LDACpin, OUTPUT); 
  digitalWrite(LDACpin, LOW);
}

void updateControl(){
  // put changing controls in here
}

void audioOutput(const AudioOutput f){ // f is a structure containing both channels
  int outL = f.l();
  int outR = f.r();
  dac.analogWrite(outL, outR, outL, outR); // write to input register of a DAC. Channel 0-3, Value 0-4095
}

AudioOutput updateAudio(){
  return MonoOutput::from8Bit(aSin.next()); // return an int signal centred around 0
}

void loop(){
  audioHook(); // required here
}
```
Thanks for your help :)

baritonomarchetto, 2024-12-22:
Hey Tom, thanks for the answer. I am using PT8211 in another project without issues, but I am not sure if it is possible to use two of them in parallel for quad output (never tested: maybe you have some clue?).
Do you eventually have a quad I2S DAC in mind that could work with Mozzi and Pi Pico before I start diggin for one? 

Thanks!

tomcombriat, 2024-12-22:
> but I am not sure if it is possible to use two of them in parallel for quad output (never tested: maybe you have some clue?).

It might be possible with the `MOZZI_OUTPUT_EXTERNAL_CUSTOM` mode… Creating two I2S instances, using one of them to trigger Mozzi to generate new samples (assuming the two dacs stay in sync), but honestly that sounds like a nightmare…

> Do you eventually have a quad I2S DAC in mind that could work with Mozzi and Pi Pico

Unfortunately, I do not have nothing in mind (I am just starting to move out from the PT8211 ;) ). Note that Mozzi does not support quad output yet ;)

baritonomarchetto, 2024-12-22:
Thank you for the help, much appreciated!

closed 2024-12-22 by baritonomarchetto