sensorium/Mozzi #300: Compile error on Arduino nano

riyant56, 2025-01-04:
On The latest version of mozzi 2.00 when i compile any code with this library it shows 
A error compiling 

/storage/emulated/0/ArduinoDroid/userlibraries/Mozzi/ hardware_defines.h:52:73: error: missing binary operator before token "("

Im doing it on Android , my pc got damaged😔
Any help is appreciated

tomcombriat, 2025-01-04:
Hi,
What board exactly are you using? I bet this isn't the old AVR based nano right?

riyant56, 2025-01-13:
Its with the ch340 ic and old bootloader

riyant56, 2025-01-14:
Any board i try it(including genuine Arduino) does not work
i think its the problem of the android app called Arduinodroid  can you help me out
Thanks for response

tomcombriat, 2025-01-14:
Hi,
I think the problem is more to be found on ArduinoDroid's side, as this works on the standard Arduino's IDE. Wild guess, I think the compiler does not set the board's define (line 24 of `hardware_defines.h`). You *could* try to add `#define __AVR__` at the very top of your sketch…

riyant56, 2025-01-14:
No luck with #define __AVR__
What can i do now 
Your help is appreciated

tomcombriat, 2025-01-14:
> What can i do now

I think your best course of action is to contact directly the authors of this app… Mozzi works with your architecture (AVR) on the standard Arduino IDE, hence is Arduino compliant but apparently the app you use is not completely and this is impossible for us, without the source code of that app (and even if we had it, would be a lot of work) to trace where is the compatibility broken…

riyant56, 2025-01-15:
Ok thanks for your response :⁠-⁠)
You helped me 
now i will try it with 
Other ide

closed 2025-01-15 by tomcombriat