sensorium/Mozzi #303: Provide a link to 1.1.2

eclab, 2025-02-01:
I have a great deal of open source firmware which does not run on 2.0, and people who use it are confused as to how to get the Mozzi 1.1.2 source code.

* The Mozzi web page, AFAIK, doesn't provide an archive for 1.1.2.
* The github repository doesn't provide a clear branch for 1.1.2 (it's not even clear what "Mozzi_1" is).
* The only known way to get 1.1.2 is to install it as a version through the IDE.

Given that 2.0 breaks 1.1.2 code, it would be helpful to provide some links to 1.1.2 repositories and source.  Else you'll start seeing my situation: providing forks and zip archives of 1.1.2, which is suboptimal for the project as a whole.

tomcombriat, 2025-02-01:
Hello,

Indeed, Mozzi 2 is not fully compatible, but if you are not using very weird config, the transition should be very easy as most of the incompatibility is on the removing of the config file. It is encouraged to stop using Mozzi 1 for further development.

As per availability:
> The Mozzi web page, AFAIK, doesn't provide an archive for 1.1.2.

It does not, but a package can be found in the releases: https://github.com/sensorium/Mozzi/releases/tag/1.1.2. This can be installed manually (copy-pasting in the library folder of Arduino), as per the way Mozzi was installed before we put it in the Arduino library repo.

> The github repository doesn't provide a clear branch for 1.1.2 (it's not even clear what "Mozzi_1" is).

There is a `Mozzi_1` on github, as stated on the website: "If desparate, there is still a “Mozzi_1” branch in the git repository". `git checkout Mozzi_1` puts you on the latest Mozzi 1 version and should be good for git users. What would you suggest?

> The only known (to me) way to get 1.1.2 is to install it as a version through the IDE.

This is fairly easy for end-users right?

What changes would you suggest?

---

>  providing forks and zip archives of 1.1.2, which is suboptimal for the Mozzi effort as a whole.

The standard way to deal with that IMO, would be for you to provide compiled packages and not only sources.

eclab, 2025-02-01:
> It does not, but a package can be found in the releases: https://github.com/sensorium/Mozzi/releases/tag/1.1.2. This can be installed manually (copy-pasting in the library folder of Arduino), as per [...]

Ah!  Okay so what I'd do is have a pointer to the 1.0.3/1.1.0/1.1.1/1.1.2 releases from the main web page, and not bother with Mozzi_1 (which I *assume* is 1.1.2?).  BTW, there's a release called x.x.x?

> > The only known (to me) way to get 1.1.2 is to install it as a version through the IDE.
> 
> This is fairly easy for end-users right?

There are two cases: using 1.1.2 as a library, and modifying or perusing the 1.1.2 code proper.  IDE installation is mostly useful  in the first case.

tomcombriat, 2025-02-01:
> Ah! Okay so what I'd do is have a pointer to the 1.0.3/1.1.0/1.1.1/1.1.2 releases from the main web page,

#304 Adds a link to the package to the last version.

> BTW, there's a release called x.x.x?

That's an old tag, probably used for testing back in the day.

> There are two cases: using 1.1.2 as a library, and modifying or perusing the 1.1.2 code proper. IDE installation is mostly useful in the first case.

Dev users will probably be okay by changing their local branch to `Mozzi_1` by `git checkout Mozzi_1` hence the necessity to have this branch. End-users still needing the v1 can use the IDE or the package. Does not that cover all use cases?

closed 2025-02-04 by tomcombriat