sensorium/Mozzi #307: Issues with ESP32-S3 using an external dac

udu3324, 2025-04-06:
Hello. I am trying to compile [this](https://github.com/sensorium/Mozzi/blob/a40a424e6fd7d1af3fe51b78dfe63f698fecb547/examples/13.External_Audio_Output/MCP4922_mono_24bits/MCP4922_mono_24bits.ino#L9) sample code, but it runs into an issue. I am using a esp32-s3 devkit board with Arduino framework and PlatformIO.

Errors below:
```
In file included from C:/Users/rice/.platformio/packages/framework-arduinoespressif32/cores/esp32/USB.h:16,
                 from C:/Users/rice/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp:14:        
C:/Users/rice/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/opi_opi/include/sdkconfig.h:143: warning: "CONFIG_BT_ENABLED" redefined
 #define CONFIG_BT_ENABLED 1

<command-line>: note: this is the location of the previous definition
In file included from C:/Users/rice/.platformio/packages/framework-arduinoespressif32/cores/esp32/USBMSC.h:18,
                 from C:/Users/rice/.platformio/packages/framework-arduinoespressif32/cores/esp32/FirmwareMSC.h:17,    
                 from C:/Users/rice/.platformio/packages/framework-arduinoespressif32/cores/esp32/FirmwareMSC.cpp:14:  
C:/Users/rice/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/opi_opi/include/sdkconfig.h:143: warning: "CONFIG_BT_ENABLED" redefined
 #define CONFIG_BT_ENABLED 1

<command-line>: note: this is the location of the previous definition
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\MD5Builder.cpp.o
In file included from lib/Mozzi/internal/MozziGuts.hpp:51,
                 from lib/Mozzi/MozziGuts.h:205,
                 from lib/Mozzi/Mozzi.h:33,
                 from src/main.cpp:36:
lib/Mozzi/internal/MozziGuts_impl_ESP32.hpp: In function 'void MozziPrivate::timer0_audio_output_isr(void*)':
lib/Mozzi/internal/MozziGuts_impl_ESP32.hpp:102:26: error: 'volatile union timg_int_clr_timers_reg_t' has no member named 'wdt'
   TIMERG0.int_clr_timers.wdt = 1;
                          ^~~
lib/Mozzi/internal/MozziGuts_impl_ESP32.hpp:103:30: error: 'volatile union timg_tnconfig_reg_t' has no member named 'en'
   TIMERG0.hw_timer[0].config.en = 1;
                              ^~
*** [.pio\build\esp32-s3-devkitc-1\src\main.cpp.o] Error 1
```
It seems like there is support for the esp32 but not the esp32-s3, which i would love for Mozzi to support fully.

tomcombriat, 2025-04-07:
Hi!

I have tried here, using the Arduino IDE and Espressif's core (https://github.com/espressif/arduino-esp32), and I get a lot of errors linked to the DAC library but not to Mozzi. Actually, commenting all DAC related instructions or compiling [this](https://github.com/sensorium/Mozzi/tree/a40a424e6fd7d1af3fe51b78dfe63f698fecb547/examples/13.External_Audio_Output/Sinewave_R2R_DAC) do work.

The port for the S3 is fairly recent (see #301), so mistakes are possible but normally it should support external audio output.

A few questions:
 - have you tried with a simpler external audio? Say, outputting on PWM?
 - what core are you using? (There is also an Arduino core for ESP32, Mozzi is not ported to that one AFAIK).
 - does the same sketch compiles with another ESP?

udu3324, 2025-04-08:
Sorry for the slow response! I just updated to the latest version of Mozzi and can finally test.

>  - have you tried with a simpler external audio? Say, outputting on PWM?

I just tried the other sketch you referenced, it compiles but does not output any frequency. It could be that my r-2r dac is using 2k/1k ohm resistors that i have on hand at the moment instead of the recommended ones.

>  - what core are you using? (There is also an Arduino core for ESP32, Mozzi is not ported to that one AFAIK).

>  - does the same sketch compiles with another ESP?

udu3324, 2025-04-13:
Got everything working perfectly! 

I had to define a few things for SPI to get the sample code up and running.
```
#define SPI_SCK   18  // SCK (Clock)
#define SPI_MOSI  13  // SDI (MOSI)
#define SPI_MISO  12  // SDO (MISO)

void setup() {
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
```

closed 2025-04-13 by udu3324

tomcombriat, 2025-04-14:
Great!